robertklep/dsm7-usb-serial-drivers #97: Failed to add USB driver to after DSM 7.2 upgrade

ryanzhengrz, 2023-06-28:
Hardware: Synology DSM920+
Firmware version: 7.2-64570 Update 1
Issue: After running the script without rebooting the Synology NAS (/usr/local/etc/rc.d/usb-serial-driveers.sh start), I checked the availability of USB driver (ls /dev/ttyU*) and it still reported "ls: cannot access '/dev/ttyU*': No such file or directory".

Background: I suspect this has anything to do with the new DSM firmware. I used to use the same approach on the same NAS with firmware version 7.1, and the USB driver was activated as expected to run a Zigbee coordinator USB dongle. Last night I upgraded the NAS firmware to the latest 7.2, and then Home Assistant cannot find the USB dongle for Zigbee anymore. I tried to duplicate my previous success by following the instructions again, but failed this time. Can someone advise what I can do next? Thanks!
[image]

Sweetgenius19, 2023-06-28:
Did you update to the 7.2 drivers? Try unplugging the device and plug it back in. Check if it is detected.

ryanzhengrz, 2023-06-29:
Yes, I updated using the 7.2 drivers for Geminilake architecture which is for DS920+. I also tried switching back to the 7.1 drivers but resulted the same. Rebooting NAS did not detect the drivers, unfortunately.

jeenyeong, 2023-06-29:
Hey, i have similar issue last night after upgrade to the latest dsm 7.2-64570 update 1. I fixed my issue after insert the driver for 7.2.

robertklep, 2023-06-29:
> After running the script without rebooting the Synology NAS (/usr/local/etc/rc.d/usb-serial-driveers.sh start), I checked the availability of USB driver (ls /dev/ttyU*) and it still reported "ls: cannot access '/dev/ttyU*': No such file or directory".

A quick way to check why this is happening is to load the modules manually, as the script will not show any errors while loading them:
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/ftdi_sio.ko
sudo insmod /lib/modules/cdc-acm.ko
sudo insmod /lib/modules/cp210x.ko
sudo insmod /lib/modules/ch341.ko
sudo insmod /lib/modules/pl2303.ko
```

My guess is that you didn't [download the modules properly](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module).

closed 2023-06-29 by robertklep