robertklep/dsm7-usb-serial-drivers #98: USB support for geminilake DSM 7.2

Masadido, 2023-07-04:
Hello,

Did anyone manage to compile and run the USB drivers for Synology DSM 7.2 Geminilake?
No success so far and the drivers I tried even managed to crash my Synology NAS :-(

Cheers,

robertklep, 2023-07-05:
Do you mean the provided drivers for [Geminilake 7.2](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2) aren't working for you?

Masadido, 2023-07-05:
I had a severe crash of my system after using those drivers. 
Not sure what happened so I would like to make sure at least someone managed to make them work before trying again...

robertklep, 2023-07-05:
But from what I understood your issue was related to the "alternative" installation method and not so much the drivers themselves?

Masadido, 2023-07-05:
Yes indeed, that's my best guess... took me 2 days to recover from last crash so I'm a bit careful now :-)

c4coer, 2023-07-07:
Good morning,
have you tested?

it works ?

Masadido, 2023-07-08:
Just a quick update. I tried again with method 1 and it seems to work this time (Geminilake drivers, DS920+, DSM 7.2).

What I did:
- Download the drivers (from this project)
- then SSH and did the following. It seems rndis_host.ko is not required.

sudo su

modprobe usbserial
modprobe ftdi_sio
modprobe cdc-acm

cd /lib/modules/

cp [sourcedirectory]/ch341.ko /lib/modules/ 
sudo insmod /lib/modules/ch341.ko

cp [sourcedirectory]/cp210x.ko /lib/modules/
sudo insmod /lib/modules/cp210x.ko

cp [sourcedirectory]/pl2303.ko /lib/modules/
sudo insmod /lib/modules/pl2303.ko

cp [sourcedirectory]/ti_usb_3410_5052.ko /lib/modules/
sudo insmod /lib/modules/ti_usb_3410_5052.ko

# Configure an autoload script
cd  /usr/local/etc/rc.d/
cp [sourcedirectory]/ti_usb_3410_5052.ko /usr/local/etc/rc.d/usb-serial-drivers.sh
sudo chmod +x usb-serial-drivers.sh

# Make sure your devices are recongnized
sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start

MPowerData, 2023-07-13:
I followed your stepts, but still am getting:

ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

Any idea on how to debug this further?

robertklep, 2023-07-13:
> ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

That means you didn't follow [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) 😋

MPowerData, 2023-07-13:
> > ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format
> 
> That means you didn't follow [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) 😋

My bad, thanks for your quick reponse! Working now. :-)

robertklep, 2023-08-29:
@tadams143 without any more information I have no idea what you're doing wrong 🤷🏼‍♂️

Are you able to load the modules without problems?

tadams143, 2023-08-29:
Sorry yeah I am not sure how to give more information, but thank you for your fast response. I have loaded all modules apart from rndis_host.ko, which it is my understanding it is unnecessary.

When I run the script it doesnt seem like anything happens, but there aren't any errors.

If i follow through the instructions again all modules say that the file exists.

robertklep, 2023-08-29:
@tadams143 that sounds like the modules loaded okay.

What does the output of this command show?
```
/usr/syno/bin/lsusb -cui
```

condoevil, 2023-09-23:
> Just a quick update. I tried again with method 1 and it seems to work this time (Geminilake drivers, DS920+, DSM 7.2).
> 
> What I did:
> 
> * Download the drivers (from this project)
> * then SSH and did the following. It seems rndis_host.ko is not required.
> 

Hi, i downloaded the files but (i'm new to linux command and issues) where should i copy those files before installation? in my PC (MAC) or on the synology?
Many thanks!

robertklep, 2023-09-23:
@condoevil they need to be installed on your Synology, so that's where they ultimately need to end up 😊

condoevil, 2023-09-23:
> @condoevil they need to be installed on your Synology, so that's where they ultimately need to end up 😊

@robertklep ok, but i actually don't know the procedure to put those files in the right place. Now i downloaded on my mac using chrome and they are in my download folder. How to proceed? Many thanks for your support, Sorry but i'm a beginner with linux things.

robertklep, 2023-09-23:
@condoevil all those commands should be executed on your Synology, which _also_ means that you need to set up SSH access to it so you can log in and execute those commands. I can't help with setting that up, but I'm sure there are plenty of tutorials to be found that explain how to get that part working.

Once you've logged in, perhaps it's easier to just download the files to your Synology directly. Which platform/Synology do you have, and which version of DSM is it running?

condoevil, 2023-09-23:
@robertklep running 7.2 and already have SSH access with SU permissions. I'm only missing all the procedure to follow next.
Synology is DS220+ and should be geminilake.

Many thanks!

robertklep, 2023-09-23:
@condoevil 

This is how you can download the files directly to DSM, from SSH:
```
cd /lib/modules
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/pl2303.ko
```

condoevil, 2023-09-23:
@robertklep Thank you!!!
after that i should proceed with the sudo insmod /lib/modules/... commands right?

robertklep, 2023-09-23:
@condoevil correct!

condoevil, 2023-09-23:
@robertklep i arrived to the last command and:
ash-4.4# sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start
/usr/local/etc/rc.d/usb-serial-drivers.sh: line 1: payload:allShortcutsEnabled:false: command not found.
What should i do?

robertklep, 2023-09-23:
@condoevil looks like you didn't download the file correctly.

Here's how to download it directly to DSM:
```
cd /usr/local/etc/rc.d
sudo wget https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/usb-serial-drivers.sh
sudo chmod 755 usb-serial-drivers.sh
```

condoevil, 2023-09-23:
@robertklep ok, now it is ok.
What should i do now? dongle working already in Home assistant o should i do something else?

robertklep, 2023-09-23:
@condoevil I assume that your goal was to get it working in HA, so goal achieved?

condoevil, 2023-09-23:
@robertklep Homeassistant is asking me for the path of the serial device. What should i insert there?

robertklep, 2023-09-23:
@condoevil are you running HA in Docker? If so, you first need to make sure the serial device gets passed to the container.

You should be able to find the device entry on the DSM side using `/usr/syno/bin/lsusb -cui`

condoevil, 2023-09-23:
@robertklep this is what i find:
ash-4.4# /usr/syno/bin/lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus c623c6b3ac29ec11a6e7bf9c40c9ce8d)
  1-2:1.0         (IF) ff:00:00 2EPs () cp210x ttyUSB0 
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500259E98EF6042)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)
ash-4.4#

robertklep, 2023-09-23:
@condoevil okay, so the device entry is `/dev/ttyUSB0`

condoevil, 2023-09-23:
@robertklep 

[image]

it says impossible to connect...

condoevil, 2023-09-23:
@robertklep 
[image]

it is listed in the /dev/ folder

robertklep, 2023-09-23:
> it says impossible to connect...

It's still not clear to me if you're using Docker, because like I said, if so, you need to pass the device to the container. I don't remember how specifically to do that on DSM since I stopped using it a while ago.

It could also be a permissions problem, try this:
```
sudo chmod 666 /dev/ttyUSB0
```

condoevil, 2023-09-23:
> It's still not clear to me if you're using Docker, because like I said, if so, you need to pass the device to the container. I don't remember how specifically to do that on DSM since I stopped using [...]

@robertklep many thanks! finally i installed MQTT and zigbee2MQTT.
Are you able to explain me how to add devices now or may i ask to someone else?

robertklep, 2023-09-23:
@condoevil you mean add devices to z2m? That's really beyond the scope of this repository, I'm sure you'll find plenty of information on the [z2m website](https://www.zigbee2mqtt.io/).

condoevil, 2023-09-23:
@robertklep Many thanks for everything! Have a nice day!!!

mvroosmalen1970, 2023-11-03:
Tried to install, but stuck at : insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format

I downloaded the modules using:
cd /lib/modules
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/pl2303.ko

Whats wrong????

robertklep, 2023-11-04:
@mvroosmalen1970 I have no idea what's wrong.

Can you post some more info:
* the output of `uname -a`
* the output of `lsusb -cui`
* the output of `dmesg | tail` right after trying to run `insmod`

Did you load `usbserial.ko` before loading the downloaded modules?

mvroosmalen1970, 2023-11-04:
> @mvroosmalen1970 I have no idea what's wrong.
> 
> Can you post some more info:
> 
> * the output of `uname -a`
> * the output of `lsusb -cui`
> * the output of `dmesg | tail` right after trying to run `insmod`
> 
> Did you load `usbserial.ko` before loading the downloaded modules?

Hi 

Solved it via: https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/

using this tutorial the USB device is recognized automatically

Mark

closed 2023-11-04 by robertklep