robertklep/dsm7-usb-serial-drivers #100: SynoCommunity related project

th0ma7, 2023-07-20:
**THIS IS NOT A BUG**

Hi, I'm one of the maintainer of SynoCommunity packages.  I've been for the most part maintaining the kernel portion of the framework.  I'm currently including DSM-7.1 & 7.2 kernel support https://github.com/SynoCommunity/spksrc/pull/5790

Noticing the overlap amongst our projects a few things comes to mind as to wether you'd be interested in participating to spksrc instead of maintaining it all on your own or ways to share knowledge amongst us.

But, on a more technical front: 
1. with 7.1 comes the arrival of `epic7002` and `rtd1619b` which uses kernel 5.x.  I wonder what you did relatively to the actual kernel configuration you've been using as it is no longer part of the source provided by Synology.  It hapens that both configuration do exist but in other arch kernel 4.4 source tree (such as `rtd1296` or any x86_64 for epic).
2. for DSM 7.2 kernel sources, have you simply faked the kernel version in the Makefile to bump it to `4.4.302` ?  As basically all older kernels didn't changed and neither did 5.5 kernels between 7.1 and 7.2, leaving only 4.4.302 missing from a kernel source perspective.

Advices on the matter would be appreciated.

You probably already know of, but just in case:
- https://github.com/SynoCommunity/spksrc
- https://synocommunity.com/package/synokernel-cdrom
- https://synocommunity.com/package/synokernel-usbserial

Also feel free to join Discord on spksrc channel where we can further discuss this if you hapen to be interested.  Link to our channels available on our frontpage https://github.com/SynoCommunity/spksrc

robertklep, 2023-07-20:
@th0ma7 I'm happy to share how I build the modules, but it's basically just all cobbled together and unstable (in the sense that compiling the modules sometimes works, sometimes not, randomly 🤷🏼‍♂️).

I'm using the module sources from the kernel source packages that _were_ made available (I believe I downloaded them from Sourceforge, can't remember) and the only thing I change between the different DSM versions are the calls to `EnvDeploy` and `PkgCreate.py` (part of the [toolkit](https://help.synology.com/developer-guide/getting_started/prepare_environment.html)). For all platforms, the same module sources seem to work on DSM 7.0, 7.1 and 7.2.

If, as you say, there are new platforms for which no sources are available, and that use different major Linux kernel versions, I doubt I can be of any help (theoretically I assume that it should be possible to download an official Linux kernel source distribution to extract the files from, provided that Synology still provides the correct build environments for `EnvDeploy` to download).

Since I don't actually use my Synology NAS anymore for USB devices (I moved to Unraid because I don't like the way Synology is stricting DSM more and more) I'm not really interested in actively maintaining these modules, at least not in the way that I think you expect.

FWIW, I'm fine with you downloading and packaging the modules in this repository if that's of any help to your project.

th0ma7, 2023-07-20:
> @th0ma7 I'm happy to share how I build the modules, but it's basically just all cobbled together and unstable (in the sense that compiling the modules sometimes works, sometimes not, randomly 🤷🏼‍♂️).

Yeah, kernel sources requires a few patches here and there to build up correctly.

> I'm using the module sources from the kernel source packages that _were_ made available (I believe I downloaded them from Sourceforge, can't remember) and the only thing I change between the [...]

This is rather interesting, I've never used that method as the spksrc framework rather downloads the toolchain to cross-compile instead of using the toolkit.  But that option is new to me, thnx for sharing.  Also, indeed Synology did cleaned-up all their sourceforge repository for older builds.  Only thing left is their own repository located at https://archive.synology.com/download

Also I find it odd that the exact same module works on any given DSM version... that's probably true for NAS based on kernel 3.x use-cases as it hasn't changed in years, but less so for 4.4 which got updated often and minor version number changed over time.

> If, as you say, there are new platforms for which no sources are available, and that use different major Linux kernel versions, I doubt I can be of any help (theoretically I assume that it should be [...]

Actually, all kernel sources are available up to DSM 7.1.  The thing is, kernels version varies from one NAS arch to the next.  As such, 3.x and 5.x linux kernels didn't changed betewen DSM 7.1 and 7.2.  Although kernel 4.4 went from `4.4.180` to `4.4.302` in the meantime, but sources for the latest version where not released.  And I did tried to port that over official kernel sources from kernel.org but there is too many synology specific changes to do that.

Also, relatively to kernel 5.x, in previous versions they used to provide the configuration file (e.g. `.config` under `synoconfigs` sub-directory).  This allowed to build any given kernel module correctly for the specific arch.  It hapens that they are not providing them anymore, although they left bread crumbles of them in kernel 4.4 sources which I can probably use to play with.

But, maybie this is where the toolkit using method you mentionned may fill this gap, something worth checking for sure.

> Since I don't actually use my Synology NAS anymore for USB devices (I moved to Unraid because I don't like the way Synology is stricting DSM more and more) I'm not really interested in actively maintaining these modules, at least not in the way that I think you expect.

gotcha

> FWIW, I'm fine with you downloading and packaging the modules in this repository if that's of any help to your project.

I don't need to.  Our spksrc framework is made to rebuild that on-demand automagically and generate distributable "easy-to-use" packages.

You may have provided the few hints I needed with the toolkit thing.  Can you guide me through how you're actually building modules using the toolkit?  I didn't see anything specific to that while reading the doc (gotta admit, read it in diagonal).

closed 2025-11-18 by robertklep