robertklep/dsm7-usb-serial-drivers #101: broadwellnk Invalid module format

mapoloco, 2023-08-01:
Hi! I follow those instructions:
https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module

and it still gives this error:

sudo -i
cd /lib/modules

rm -r ch341.ko
rm -r cp210x.ko
rm -r pl2303.ko
rm -r rndis_host.ko
tm -r ti_usb_3410_5052.ko

sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/pl2303.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/rndis_host.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/ti_usb_3410_5052.ko

sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/usb-serial-drivers.sh
cp -r usb-serial-drivers.sh /usr/local/etc/rc.d/
chmod +x /usr/local/etc/rc.d/usb-serial-drivers.sh

sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/cp210x.ko
sudo insmod /lib/modules/ch341.ko
sudo /sbin/modprobe usbserial
sudo /sbin/modprobe ftdi_sio
sudo /sbin/modprobe cdc-acm
sudo chmod 666 /dev/ttyACM*
sudo chmod 666 /dev/ttyUSB*

And I get those errors:

root@:/lib/modules# sudo insmod /lib/modules/usbserial.ko
insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists
root@:/lib/modules# sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format
root@:/lib/modules# sudo insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format
root@:/lib/modules# sudo /sbin/modprobe usbserial
root@:/lib/modules# sudo /sbin/modprobe ftdi_sio
root@:/lib/modules# sudo /sbin/modprobe cdc-acm
root@:/lib/modules# sudo chmod 666 /dev/ttyACM*
chmod: cannot access '/dev/ttyACM*': No such file or directory
root@:/lib/modules# sudo chmod 666 /dev/ttyUSB*
chmod: cannot access '/dev/ttyUSB*': No such file or directory

How can I get a valid module format ?
Thank you very much.

robertklep, 2023-08-01:
> How can I get a valid module format ?

By reading [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) carefully!

closed 2023-08-01 by robertklep

robertklep, 2023-08-01:
Open the link that you're trying to download in a browser and see if your browser downloads the module, or if you're getting a regular web page (spoiler: you're getting a regular web page).

You need to download the link below the download button:
[image]

mapoloco, 2023-08-01:
ok, thank you very much dear