robertklep/dsm7-usb-serial-drivers #103: v1000 DSM7.2 - cp210x wrong version

robotictacos, 2023-08-09:
I explicitly RTFM and downloaded the drivers via both git and also tried the download link on the webpage.  When I insmod, I get a "Invalid Module Format" error.  When I run dmesg, I see this entry: "cp210x: version magic '4.4.180+ SMP mod_unload ' should be '4.4.302+ SMP mod_unload '"

I am able to insmod the other driver files listed for v1000, just not the cp210x.ko, which unfortunately is the one I need.  I tried building the driver myself, but I cannot get it to compile.  FWIW, I am building on my work laptop running the latest Arch (btw).

Here is an excerpt of the python error messages:

`Traceback (most recent call last):
  File "/home/dave/toolkit/pkgscripts-ng/PkgCreate.py", line 184, in <module>
    main(sys.argv[1:])
  File "/home/dave/toolkit/pkgscripts-ng/PkgCreate.py", line 177, in main
    packer.pack_package()
  File "/home/dave/toolkit/pkgscripts-ng/include/python/pkguniform.py", line 437, in pack_package
    worker.execute()
  File "/home/dave/toolkit/pkgscripts-ng/include/python/pkgcommon.py", line 86, in execute
    self._process_output(self._run(*argv))
                         ^^^^^^^^^^^^^^^^
  File "/home/dave/toolkit/pkgscripts-ng/include/python/pkgcommon.py", line 212, in _run
    return doPlatformParallel(self.run_command, self.env_config.platforms)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dave/toolkit/pkgscripts-ng/include/python/parallel.py", line 60, in doPlatformParallel
    output[item] = results[item].get()
                   ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/pool.py", line 774, in get
    raise self._value
exec_env.EnvError: ('Execute chroot /home/dave/toolkit/build_env/ds.v1000-7.2 env PackageName=4.4.x /pkgscripts-ng/SynoBuild --v1000 -c --min-sdk 7.2 --parallel 20 4.4.x failed', 'Set cache size limit to 8.0 GB\nStatistics zeroed\n[INFO] projectList=4.4.x\n[INFO] Start to build 4.4.x.\n[SCRIPT] build script: //source/4.4.x/SynoBuildConf/build\n[INFO] ======= Run build script =======\nrm -rf *.o ch341.ko cp210x.ko rndis_host.ko pl2303.ko ti_usb_3410_5052.ko *.cmd\nmake -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules\nmake -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules\nmake -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules\nmake -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules\nmake -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules\nmake[1]: warning: jobserver unavailable: using -j1.  Add \'+\' to parent make rule.\nmake[1]: warning: jobserver unavailable: using -j1.  Add \'+\' to parent make rule.\nmake[1]: Entering directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nmake[1]: warning: jobserver unavailable: using -j1.  Add \'+\' to parent make rule.\nmake[1]: Entering directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nmake[1]: warning: jobserver unavailable: using -j1.  Add \'+\' to parent make rule.\nmake[1]: warning: jobserver unavailable: using -j1.  Add \'+\' to parent make rule.\nmake[1]: Entering directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nmake[1]: Entering directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nmake[1]: Entering directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\n  CC [M]  //source/4.4.x/ch341.o\n  CC [M]  //source/4.4.x/ch341.o\n  CC [M]  //source/4.4.x/ch341.o\n  CC [M]  //source/4.4.x/ch341.o\n  CC [M]  //source/4.4.x/ch341.o\nfixdep: //source/4.4.x/.ch341.o.d is empty\nfixdep: //source/4.4.x/.ch341.o.d is empty\nmv: cannot stat \'//source/4.4.x/.ch341.o.tmp\': No such file or directory\nscripts/Makefile.build:286: recipe for target \'//source/4.4.x/ch341.o\' failed\nmake[2]: *** [//source/4.4.x/ch341.o] Error 1\nmake[2]: *** Deleting file \'//source/4.4.x/ch341.o\'\nMakefile:1467: recipe for target \'_module_//source/4.4.x\' failed\nmake[1]: *** [_module_//source/4.4.x] Error 2\nmake[1]: Leaving directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nMakefile:8: recipe for target \'ti_usb_3410_5052.ko\' failed\nmake: *** [ti_usb_3410_5052.ko] Error 2\nmake: *** Waiting for unfinished jobs....\nfixdep: error opening depfile: //source/4.4.x/.ch341.o.d: No such file or directory\nscripts/Makefile.build:286: recipe for target \'//source/4.4.x/ch341.o\' failed\nmake[2]: *** [//source/4.4.x/ch341.o] Error 2\n  CC [M]  //source/4.4.x/cp210x.o\nMakefile:1467: recipe for target \'_module_//source/4.4.x\' failed\nmake[1]: *** [_module_//source/4.4.x] Error 2\nmake[1]: Leaving directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nMakefile:8: recipe for target \'ch341.ko\' failed\nmake: *** [ch341.ko] Error 2\nfixdep: error opening depfile: //source/4.4.x/.ch341.o.d: No such file or directory\nscripts/Makefile.build:286: recipe for target \'//source/4.4.x/ch341.o\' failed\nmake[2]: *** [//source/4.4.x/ch341.o] Error 2\nMakefile:1467: recipe for target \'_module_//source/4.4.x\' failed\nmake[1]: *** [_module_//source/4.4.x] Error 2\nmake[1]: Leaving directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nMakefile:8: recipe for target \'pl2303.ko\' failed\nmake: *** [pl2303.ko] Error 2\nfixdep: error opening depfile: //source/4.4.x/.ch341.o.d: No such file or directory\nscripts/Makefile.build:286: recipe for target \'//source/4.4.x/ch341.o\' failed\nmake[2]: *** [//source/4.4.x/ch341.o] Error 2\nMakefile:1467: recipe for target \'_module_//source/4.4.x\' failed\nmake[1]: *** [_module_//source/4.4.x] Error 2\nmake[1]: Leaving directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nMakefile:8: recipe for target \'rndis_host.ko\' failed\nmake: *** [rndis_host.ko] Error 2\n  CC [M]  //source/4.4.x/rndis_host.o\n  CC [M]  //source/4.4.x/pl2303.o\n  CC [M]  //source/4.4.x/ti_usb_3410_5052.o\n  Building modules, stage 2.\n  MODPOST 4 modules\n//source/4.4.x/ch341.o: No such file or directory\nscripts/Makefile.modpost:91: recipe for target \'__modpost\' failed\nmake[2]: *** [__modpost] Error 1\nMakefile:1470: recipe for target \'modules\' failed\nmake[1]: *** [modules] Error 2\nmake[1]: Leaving directory \'/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build\'\nMakefile:8: recipe for target \'cp210x.ko\' failed\nmake: *** [cp210x.ko] Error 2\n[Error] Build project fail!\nTime cost: 00:00:01 [Build-->4.4.x]\n[INFO] Build 4.4.x finished!\n\n----------------- Time cost statistics -----------------\nTime cost: 00:00:01 [Build-->4.4.x]\n\n########################################################\n      Error(s) occurred on project "4.4.x"\n########################################################\n1 projects, 1 failed, 0 blocked.\n\n[Error] Check [/logs/error.build] for fixing errors.')
dave@arch-xps17 ~/t/s/4.4.x [1]> 
`

robertklep, 2023-08-10:
> When I run dmesg, I see this entry: "cp210x: version magic '4.4.180+ SMP mod_unload ' should be '4.4.302+ SMP mod_unload '"

It looks like you downloaded the DSM 7.1 version of that module:
```
dsm-7.1/cp210x.ko: vermagic=4.4.180+ SMP mod_unload
dsm-7.2/cp210x.ko: vermagic=4.4.302+ SMP mod_unload
```

robotictacos, 2023-08-10:
I am quite certain I was in the 7.2 directory of this project, I did a git clone of the whole project.

robertklep, 2023-08-10:
What does this command output?
```
grep -ao 'vermagic=[0-9.]\+' cp210x.ko
```

robotictacos, 2023-08-10:
'sh-4.4$ grep -ao 'vermagic=[0-9.]\+' cp210x.ko'
'vermagic=4.4.302'

robotictacos, 2023-08-10:
I am good now.  The issue is that my script used the cp -n command which left the old driver file in place, so when I was running insmod it was against the 7.0 version of the driver not the 7.2 version.  Thanks for your help though, your advice to run that grep command is what led me to the solution.

closed 2023-08-10 by robotictacos