robertklep/dsm7-usb-serial-drivers #104: USB device not found after reboot DSM 7.2

chinswain, 2023-09-23:
Hi all,

I sucessfully connected a Sonoff Zigbee 3.0 USB stick to my Synology DS818+ running DSM 7.2 and set the below as a scheduled task to run on startup:

```
/sbin/modprobe usbserial
/sbin/modprobe ftdi_sio
/sbin/modprobe cdc-acm
chmod 777 /dev/ttyUSB0
chmod 777 /dev/ttyACM0
```
However, now my coral.ai edge TPU USB does not connect when restarting the synology. It looks like the device doesn't exist until it's removed and pluged back in. Does anyone have any advice on getting it to mount? Anything I can add to the startup script?

lsusb after a reboot:
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0bda:5411:0004 09  2.10  480MBit/s 0mA 1IF  (Generic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-1.3     06da:ffff:0003 00  1.10  1.5MBit/s 100mA 1IF  (PPC Offline UPS 000000000   )
  |__1-3         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0088906EDA5A57)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:0411:0004 09  3.20 5000MBit/s 0mA 1IF  (Generic USB3.2 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
```
Device physically removed and pluged back in (2-2.4):
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0bda:5411:0004 09  2.10  480MBit/s 0mA 1IF  (Generic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-1.3     06da:ffff:0003 00  1.10  1.5MBit/s 100mA 1IF  (PPC Offline UPS 000000000   )
  |__1-3         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0088906EDA5A57)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:0411:0004 09  3.20 5000MBit/s 0mA 1IF  (Generic USB3.2 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
    |__2-2.4     1a6e:089a:0100 00  3.10 5000MBit/s 896mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad)
```

I'm mounting this device to a --privileged docker container with --device /dev/bus/usb:/dev/bus/usb \
The Zigbee stick is connecting fine after a reboot.

robertklep, 2023-09-24:
Are you using external USB hubs? Might be a power issue, although from what it looks like you're using different ports for the Sonoff and the Coral.

chinswain, 2023-09-24:
Yes - I have added a powered USB 4 port hub, hoping that would help. The Sonoff Zigbee device is in the front USB port and the Coral is in the back (via the port). The only other device is the UPS which is picked up ok.

robertklep, 2023-09-24:
It might be related to the Coral, see this very similar issue: https://github.com/home-assistant/operating-system/issues/2349

Perhaps it'll work if you shuffle the devices around a bit.

RKADD, 2023-10-25:
I had a Zigbee adapter on the USB port on the DS220 and it worked until I had to restart the NAS today. I've tried many things, but in the end /sbin/modprobe usbserial always gives the error "could not insert 'usbserial': Exec format error". The ko drivers are still the same. Rights are also true. Is anyone able to help me?

closed 2023-10-25 by chinswain

robertklep, 2023-10-26:
> "could not insert 'usbserial': Exec format error"

It sounds like you overwrote `/lib/modules/usbserial.ko` with something else. You'll have to see if you can find it from somewhere else.

RKADD, 2023-10-26:
Yes, I installed the drivers for the first time in July and it worked. I now had the ch341 again, but it was still the same.

-rw-r--r-- 1 root root  47963 Jun  2 20:20 /lib/modules/usbserial_1.ko
-rw-r--r-- 1 root root 984784 Jul 10 21:28 /lib/modules/usbserial.ko
sh-4.4# ls -l /lib/modules/ch*
-rw-r--r-- 1 root root 357080 Jul 10 21:28 /lib/modules/ch341_2.ko
-rw-r--r-- 1 root root 255928 Jul 10 21:55 /lib/modules/ch341.ko.1
-rw-r--r-- 1 root root 255928 Oct 25 22:45 /lib/modules/ch341.ko

The following packages are still installed and up-to-date on the NAS: SyncoKernel USB, SyncoCLI Kernel Tools, etc.

robertklep, 2023-10-26:
> -rw-r--r-- 1 root root 47963 Jun 2 20:20 /lib/modules/usbserial_1.ko
> -rw-r--r-- 1 root root 984784 Jul 10 21:28 /lib/modules/usbserial.ko

My guess is that the first file is the actual module, I don't know where the second file is coming from but the error points to it not being a module at all.

RKADD, 2023-10-26:
The second File comes from SyncoKernel USB Paket.
I copied it there and that's how it worked initially

cp /volume1/@appstore/synokernel-usbserial/lib/modules/geminilake-7.0/4.4.180+/drivers/usb/serial/* /lib/modules/

robertklep, 2023-10-26:
All I can say is that it's not working because `/lib/modules/usbserial.ko` isn't a valid module. I don't know why you would need a module from the SynoKernel package if your NAS comes with a working one, but if you want this fixed, you're going to have to take this up with the people responsible for that package.

The only suggestion I can make is to copy the old file (`usbserial_1.ko`) back because that will very likely fix your problem.

RKADD, 2023-10-26:
Perfect, that was it. I reactivated the old drivers and it works now. I don't understand it, but that doesn't matter right now. Thank you.