robertklep/dsm7-usb-serial-drivers #105: Invalid module format  - DSM7.2 - synology_braswell_716+

jujuinparis, 2023-10-02:
Hi

First, thanks a lot for your precious work 

I have finally decided to upgrade my old DS716 to DSM7.2 and I am struggling with USB drivers.
I managed to get ttyACM0 and ttyACM1 working but I can't get my bluetooth and USB Esp to get recognized

I have followed your instructions 
```

sudo su

modprobe usbserial
modprobe ftdi_sio
modprobe cdc-acm

cd /lib/modules/

sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/braswell/dsm-7.2/ch341.ko
sudo insmod /lib/modules/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/braswell/dsm-7.2/cp210x.ko
sudo insmod /lib/modules/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/braswell/dsm-7.2/pl2303.ko
sudo insmod /lib/modules/pl2303.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/braswell/dsm-7.2/ti_usb_3410_5052.ko
sudo insmod /lib/modules/ti_usb_3410_5052.ko

# Configure an autoload script

cd  /usr/local/etc/rc.d/
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/usb-serial-drivers.sh
sudo chmod +x usb-serial-drivers.sh

# Make sure your devices are recongnized

sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start
```

but I am always getting the below for all modules 

insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format

can you please kindly help ?

thanks

robertklep, 2023-10-02:
Those aren't my instructions, you're downloading from the wrong URL 🥴

jujuinparis, 2023-10-02:
Indeed !!  https://github.com/robertklep/dsm7-usb-serial-drivers/**blob/main**/modules/geminilake/dsm-7.1/ch341.ko is NOT CORRECT   

I need the RAW URL https://github.com/robertklep/dsm7-usb-serial-drivers/**raw**/main/modules/braswell/dsm-7.1/ch341.ko

and that works :D  

thanks mate ! (maybe indicate this trick in the readme)

robertklep, 2023-10-02:
The README contains just that, but apparently not clear enough 😉

closed 2023-10-02 by robertklep