robertklep/dsm7-usb-serial-drivers #106: USB support for "Sonoff Zigbee 3.0 USB Dongle Plus V2" & geminilake DSM 7.2

steph54000, 2023-10-10:
Hi,
I'm new to home automation and Linux...
I have a Synology DS220+ geminilake DSM 7.2-64570.
I bought : “Sonoff Zigbee 3.0 USB Dongle Plus V2” model "ZBDongle-E" 
	- Chip : CH9102/CH343 (not CC2652P)
	- Silicon Labs EFR32MG21
	- EmberZNet (EZSP v8)
	- pre-flashing with coordinator EmberZNet 6.10.3 firmware
	
I followed your explanations here : [https://github.com/robertklep/dsm7-usb-serial-drivers/issues/98](url)

to sum up : 
- Access with PuTTY
sudo su

modprobe usbserial
modprobe ftdi_sio
modprobe cdc-acm

cd /lib/modules/

sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko
sudo insmod /lib/modules/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko
sudo insmod /lib/modules/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/pl2303.ko
sudo insmod /lib/modules/pl2303.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ti_usb_3410_5052.ko
sudo insmod /lib/modules/ti_usb_3410_5052.ko

cd /usr/local/etc/rc.d/
sudo wget https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/usb-serial-drivers.sh
sudo chmod 755 usb-serial-drivers.sh

sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start

And finaly
$ /usr/syno/bin/lsusb -cui
|__usb1 1d6b:0002:0404 09 2.00 480MBit/s 0mA 1IF (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
|__1-4 f400:f400:0100 00 2.00 480MBit/s 200mA 1IF (Synology DiskStation 7F001154D4CC4585)
1-4:1.0 (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)

I guess I don't have the good driver "CH9102.ko or CH343.ko" but I don't understand what to do here: [https://github.com/WCHSoftGroup/ch343ser_linux](url)

To be sure of the key, I flashed again in coordinator EmberZNet 6.10.3 firmware but it's the same.

Can anyone help me ?
Thanks

robertklep, 2023-10-10:
It should work with the ch341.ko driver. Are you sure it's loaded correctly? What does `lsmod -a | grep serial` show?

robertklep, 2023-10-10:
Oh never mind, it looks like the Synology cannot even see your device, it should be shown in the output of `lsusb` but it isn't.

steph54000, 2023-10-10:
Hi,
dmesg | grep -i serial
[    0.000000] Command line: root=/dev/md0 intel_iommu=igfx_off netif_num=2 HddEnableDynamicPower=1 syno_hw_version=DS220+ vender_format_version=2 synoboot                      2 sn=2290RLR7X4CAV macs=9009d030a827,9009d030a828 syno_ttyS0=pciserial,0x0:0x18.0x2,115200 syno_ttyS1=pciserial,0x0:0x18.0x0,115200
[    0.000000] Kernel command line: root=/dev/md0 intel_iommu=igfx_off netif_num=2 HddEnableDynamicPower=1 syno_hw_version=DS220+ vender_format_version=2 s                      ynoboot2 sn=2290RLR7X4CAV macs=9009d030a827,9009d030a828 syno_ttyS0=pciserial,0x0:0x18.0x2,115200 syno_ttyS1=pciserial,0x0:0x18.0x0,115200
[    0.000000] Serial Number: 2290RLR7X4CAV
[    5.017685] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
[   42.425537] systemd[1]: Created slice synokernel-usbserial's slice.
[   42.438651] systemd[1]: Starting synokernel-usbserial's slice.
[   43.186739] systemd[1]: Created slice system-serial\x2dgetty.slice.
[   43.199874] systemd[1]: Starting system-serial\x2dgetty.slice.
[   68.178569] usbcore: registered new interface driver usbserial
[   68.210709] usbserial: USB Serial support registered for FTDI USB Serial Device
[   68.312563] usbserial: USB Serial support registered for cp210x
[   68.367815] usbserial: USB Serial support registered for ch341-uart
[   68.552545] usbserial: USB Serial support registered for pl2303
[   68.826786] usbserial: USB Serial support registered for TI USB 3410 1 port adapter
[   68.835360] usbserial: USB Serial support registered for TI USB 5052 2 port adapter

steph54000, 2023-10-10:
lsmod | grep ch341 

ch341                   5967  0
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  17 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

steph54000, 2023-10-10:
lsmod | grep serial

usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  17 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

robertklep, 2023-10-10:
The required modules are loaded, so this looks to be a hardware issue 😢 Do you have a PC or Mac that you can plug the Sonoff dongle in to to see if anything happens there?

steph54000, 2023-10-11:
hi,
on my PC :
[image]
[image]

And, to be sure, i I flashed it in coordinator EmberZNet 6.10.3 firmware but it's the same

steph54000, 2023-10-11:
Maybe a CH343 pilot ?

robertklep, 2023-10-12:
As I explained: your Synology doesn't even see the device is plugged in, otherwise it would have shown up in the output of `lsusb`. That's a hardware issue.

Try using a USB extension cord to see if that makes it show up.

steph54000, 2023-10-12:
It’s miraculous, it works!
The dongle plugged directly into the NAS is not recognized.
I just added a USB extension, everything works.
Many thanks my friend!
[image]

robertklep, 2023-10-12:
Have fun!

closed 2023-10-12 by robertklep