robertklep/dsm7-usb-serial-drivers #107: ConBee II not found on DS723+ (R1000)

pauliivan93, 2023-10-13:
**Type of device**
What type of device (make, model, USB serial chip if you know it) are you trying to add support for? (Please be aware that the modules in the repository are only for USB-to-serial devices, not for Bluetooth, TV receivers, audio interfaces, etc)

**Have the modules loaded?**
Are you sure that the modules you downloaded from this repository have loaded properly? To check, you can run the following commands (after downloading/installing the modules):
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/ftdi_sio.ko
sudo insmod /lib/modules/cdc-acm.ko
sudo insmod /lib/modules/cp210x.ko
sudo insmod /lib/modules/ch341.ko
sudo insmod /lib/modules/pl2303.ko
```
These commands shouldn't return any errors. If they do, please post them here (unless the error is _"Invalid module format"_, in which case you should [follow the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) carefully).

**Output of lsusb**
Please include the output of the following command:
```
/usr/syno/bin/lsusb -cui
```

```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500320F9481C669)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host7 (synoboot)
  |__1-2         1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (dresden elektronik ingenieurtechnik GmbH ConBee II DE2437393)
  1-2:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0 
  1-2:1.1         (IF) 0a:00:00 2EPs () cdc_acm 
```

robertklep, 2023-10-14:
@pauliivan93 what's the problem you're having, though? Your Conbee is being recognized properly, according to the output of `lsusb` 😄

pauliivan93, 2023-10-14:
Im sorry meanwhile everything works, sometimes it found the Conbee 2 and sometimes not. Just need to findout why no Zigbee devices found. So we can close the issue.

robertklep, 2023-10-14:
@pauliivan93 try using an extension cord between your Synology and the Conbee.

closed 2023-10-14 by robertklep

pauliivan93, 2023-10-14:
Yeah thank you, i will try new stick "Sky connect" with extension, should be here in some days.

robertklep, 2023-10-14:
I've used the Conbee for quite some time on Synology and it worked fine, but only with an extension cord. USB 3 ports can cause a lot of interference in the Zigbee (and WiFi) frequency range.

pauliivan93, 2023-10-14:
For some reason the zha in HA doesnt see new pairing devices, so im hope the sky connect would be working better.