robertklep/dsm7-usb-serial-drivers #108: USB support for rtd1619b DSM 7.2 is not present

Riqardos, 2023-10-28:
Currently running DS423 with DSM 7.2, but can't find/install correct usb drivers.
Is there a plan to support those drivers as well? Have they already been made available for download somewhere?

Thanks

robertklep, 2023-10-29:
I can have a look and see if I can get the modules to compile for that platform. Can you tell me which kernel version it's running (to find out, run `uname -a` from a terminal)?

enricocirignaco, 2023-11-03:
I have the same problem here with my ds223
That's the output for **uname -a**:  ```Linux Home-NAS 5.10.55+ #64570 SMP Thu Jul 20 00:06:13 CST 2023 aarch64 GNU/Linux synology_rtd1619b_ds223```
Your help would be much appreciated!

robertklep, 2023-11-03:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1619b/dsm-7.2

(please [follow the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main#downloading-a-module))

enricocirignaco, 2023-11-11:
Sorry for the late response. I tried the new drivers you added and they works for me! Many Thanks, I appreciate your work!
I will list some information about my system in case someone has to troubleshoot:
- HW: Synology DS223
- CPU Architecture: rtd1619b
- OS: DSM 7.2-64570 Update 3
- Tested with: Zigbee Dongle EFR32MG21 chip with CH340 serial adapter

robertklep, 2023-11-11:
Thanks for testing and reporting back! I'll close this issue for now.

closed 2023-11-11 by robertklep