robertklep/dsm7-usb-serial-drivers #109: Unknown symbol V1000 DS1821+

bertybassett, 2023-10-31:
Synology Model DS1821+
CPU v1000

DS1821+
`
root@SynologyNAS:/lib/modules# sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/ch341.ko
--2023-10-31 21:06:12--  https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/ch341.ko
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/ch341.ko [following]
--2023-10-31 21:06:12--  https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/ch341.ko
Resolving raw.githubusercontent.com... 185.199.109.133, 185.199.108.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 255928 (250K) [application/octet-stream]
Saving to: 'ch341.ko.1'

ch341.ko.1                        100%[=============================================================>] 249.93K  --.-KB/s    in 0.03s   

2023-10-31 21:06:13 (7.21 MB/s) - 'ch341.ko.1' saved [255928/255928]

root@SynologyNAS:/lib/modules# ls ch*
ch341.ko  ch341.ko.1
root@SynologyNAS:/lib/modules# rm ch341.ko.1
root@SynologyNAS:/lib/modules# sudo insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Unknown symbol in module
`

[image]

bertybassett, 2023-10-31:
noticed this could it be something?

root@SynologyNAS:/lib/modules# grep -ao 'vermagic=[0-9.]\+' usbserial.ko 
vermagic=4.4.180

robertklep, 2023-11-01:
Did you actually load `usbserial.ko` before the others?

bertybassett, 2023-11-01:
I think I over write the usbserial.ko with a file from somewhere else. 

This was working last week using the  files from here and the .sh script then I upgraded dms a few spine points and additional the .ko files were missing. 

I quickly re-copied what I thought was the 7.2 .ko files back to lib\modules but I think I copied a rouge 7.0 usbserial.ko as well. 

Any idea who I test if my usbserial.ko is from 7.0?  

might log a support call with Synology support and see if they will send me a 7.2 usbserial file. 

Does compiling the files by hand create the usbserial.ko.

robertklep, 2023-11-01:
What happens if you try to load `usbserial.ko`?

Compiling manually will not create `usbserial.ko`, because it's already supplied by DSM itself.

bertybassett, 2023-11-01:
reboot just because then

[image]

robertklep, 2023-11-01:
Yes, that looks like it has been overwritten 🙁 It's not likely to be a version for an older kernel, because that would yield a different error.

You need to find someone that has the same NAS (or at least the same platform) to provide you with the correct copy of that file, or (possibly) download your current firmware from Synology and try to install it again.

bertybassett, 2023-11-01:
dmesg confirm my thoughts I think

[image]

robertklep, 2023-11-01:
Your screenshot isn't showing 😅

bertybassett, 2023-11-01:
[image]

bertybassett, 2023-11-01:
yes we are on the same page ,99% sure usbserial.ko is from DSM 7.0 and needs to be from 7.2

bertybassett, 2023-11-01:
yeap was the wrong version of usbserial.ko

installed xpnology v1000 and extracted a working file, re-imported and rebooted real sysnology and ttyUSB0 appeared

robertklep, 2023-11-01:
Ah, good thinking! Glad to see it's solved 👍🏻

closed 2023-11-01 by robertklep