robertklep/dsm7-usb-serial-drivers #110: Unable to install usb drivers on dsm 7.2 - Invalid module format

jakzim87, 2023-11-06:
Hi all,
i have followed all the instructions and wanted to install usb drivers for a usb zigbee devices on my synology. (No ttyusb or ttyacm device in /dev)

Unfortunately i am running dsm 7.2 (Linux MyCloud 4.4.302+ #64570 SMP Tue Jun 13 21:39:55 CST 2023 x86_64 GNU/Linux synology_geminilake_720+)

It is a 720+ geminilake synology. I did try these drives here: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2

None of those i could install with this command as described in the manual: insmod /lib/modules/cp210x.ko
I am always getting this error: 

sudo insmod /lib/modules/cp210x.ko 
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

Any help?

robertklep, 2023-11-06:
> None of those i could install with this command as described in the manual

Which manual is that?

> Invalid module format

That's caused by not downloading the correct files.

jakzim87, 2023-11-06:
Hey @robertklep and thanks for this quick reply.

I basically ran the modprobe commands and then wget from your page here (link mentioned above, https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2)

Then confirm the files were downlaoded into /lib/modules and then run the insmod with the error as outcome as said.

Which files would be the right ones? Can you point me to those? I checked on the page you mentioned (https://kb.synology.com/en-uk/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have) and found my 720+ as geminilake and then used the 7.2 geminilake folder and files here: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2

Thanks again

arajeet, 2023-11-06:
Even i am facing the same issue. I have ds220j . When i did the below  and tried insmod,  i am getting invalid format.  
I am on  DSM 7.2-64570 Update 3. Is there some issue

wget https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1296/dsm-7.2/cp210x.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1296/dsm-7.2/ch341.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1296/dsm-7.2/pl2303.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1296/dsm-7.2/rndis_host.ko

robertklep, 2023-11-07:
Please make sure that you follow [the correct downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module)!

jakzim87, 2023-11-07:
Hey Robert,

thanks again for your reply and help! That was helpful as it wasnt an obvious issue for me. It did in fact download correctly using the other link so i didnt think it was a problem. But actually it was, so to other who may also be experiencing the smae issue:

I have used this link: https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/geminilake/dsm-7.2/ch341.ko

But this is wrong and you actually need to use:
https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko

Quite confusing because both actually download a file with the same name
Thanks!

arajeet, 2023-11-07:
Hi Robert,

Sorry not sure what i am doing wrong.
root@araku:/lib/modules# uname -a
Linux araku 4.4.302+ #64570 SMP Thu Jul 20 00:06:53 CST 2023 aarch64 GNU/Linux synology_rtd1296_ds220j
root@araku:/lib/modules# wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/rtd1296/dsm-7.2/ch341.ko

root@araku:/lib/modules# insmod ch341.ko 
insmod: ERROR: could not insert module ch341.ko: Unknown symbol in module
root@araku:/lib/modules# pwd

Could you pleae adivse

arajeet, 2023-11-07:
root@araku:/lib/modules# insmod  ch341.ko 
insmod: ERROR: could not insert module ch341.ko: Unknown symbol in module
root@araku:/lib/modules# insmod  ti_usb_3410_5052.ko 
insmod: ERROR: could not insert module ti_usb_3410_5052.ko: Unknown symbol in module
root@araku:/lib/modules# insmod ti_usb_3410_5052.ko 
insmod: ERROR: could not insert module ti_usb_3410_5052.ko: Unknown symbol in module
root@araku:/lib/modules# insmod rndis_host.ko 
insmod: ERROR: could not insert module rndis_host.ko: Unknown symbol in module
root@araku:/lib/modules# insmod pl2303.ko 
insmod: ERROR: could not insert module pl2303.ko: Unknown symbol in module
root@araku:/lib/modules# insmod cp210x.ko 
insmod: ERROR: could not insert module cp210x.ko: Unknown symbol in module
root@araku:/lib/modules# pwd
/lib/modules
root@araku:/lib/modules# 

all the files same issue

robertklep, 2023-11-08:
@arajeet you didn't load `usbserial.ko` first.

c4coer, 2023-11-18:
hello every body, for me, no problem update 7.1 to 7.2, and 7.2 to DSM 7.2.1-69057 Update 1 for Ds720plus

Lars-B84, 2023-11-20:
> hello every body, for me, no problem update 7.1 to 7.2, and 7.2 to DSM 7.2.1-69057 Update 1 for Ds720plus

Thanks!  I was looking for this news ;)

brrrgh, 2023-11-23:
Hi Robert,
seems like I have similar problem but just for the rndis_host.ko file:
```
/lib/modules$ sudo insmod /lib/modules/rndis_host.ko
insmod: ERROR: could not insert module /lib/modules/rndis_host.ko: Unknown symbol in module
```

What does this mean? I didn’t find any usbserial.ko file anywhere?

> @arajeet you didn't load `usbserial.ko` first.

robertklep, 2023-11-23:
@brrrgh do you actually need `rndis_host.ko`?

The full path is `/lib/modules/usbserial.ko`, and it should come with DSM. [The provided script](https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/usb-serial-drivers.sh) will load it, so if you don't use that script you'll need to load it manually _before_ any of the modules from this repository.

brrrgh, 2023-11-23:
OK, got it. `usbserial.ko` is already there and I don’t probably need `ndis_host.ko`.
But how to check if the drivers were loaded after running `# /usr/local/etc/rc.d/usb-serial-drivers.sh start`?
I though it will be listed under `/dev` as `ttyUSB0`?

robertklep, 2023-11-23:
To check if the drivers are loaded:
```
lsmod | grep serial
```

To check if your device was recognised, and which device entry it uses:
```
lsusb -cui
```

brrrgh, 2023-11-23:
wohoo, you’re my hero!
Thanks a lot

theorthoteam, 2023-12-04:
thanks a lot @robertklep, I attached a usb hub and I connected the USB ZibBee pen on it. It worked! 
For some reason the kernel couldn't assign an id to the Zigbee Antenna but with usb hub  kernel got that id. Anyway it worked!

You are a superhero ;-)

spock99, 2024-02-13:
Does anyone have bluetooth drivers for 7.2.x and Apollolake DSM 918+
I see it see the USB device
lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         051d:0002:0106 00  1.10  1.5MBit/s 2mA 1IF  (APC Back-UPS ES 650G1 FW:906.W1 .D USB FW:W1  4B2103P14746  )
  1-1:1.0         (IF) 03:00:00 1EP  () usbfs
  **|__1-3         2357:0604:0200 e0  1.10   12MBit/s 500mA 2IFs (  TP-Link UB500 Adapter E848B8C82000)**
  1-3:1.0         (IF) e0:01:01 3EPs ()
  1-3:1.1         (IF) e0:01:01 2EPs ()
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008C6930DA5A89)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host5 (synoboot)

But the drivers are not for correct CPU 
/lib/modules# sudo insmod /lib/modules/bluetooth.ko
insmod: ERROR: could not insert module /lib/modules/bluetooth.ko: Invalid module format
/lib/modules# sudo insmod /lib/modules/btusb.ko
insmod: ERROR: could not insert module /lib/modules/btusb.ko: Invalid module format

These are geminilake which as I found out do not work with Apollolake.
Can anyone compile the correct ones for me?

Thanks
Dan

stagovic, 2024-02-23:
Hi all,
After having had some trouble with my sonoff zigbee stick after the update of my synology DS420+ to version 7.2 these could be resolved with using the drivers from "the raw folder".
Obviously I felt a little bit too safe as I run into the "insmod" problem again after having updated to DSM 7.2.1-69057 Update 4.
Does anyone have an idea why I get "insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format" again and how to resolve this? Thanks in advance for any help.

I currently use these drivers:
   https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko
   https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko
   https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/pl2303.ko
   https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ti_usb_3410_5052.ko

robertklep, 2024-02-23:
@stagovic the size of `/lib/modules/ch341.ko` should be 255928 bytes, and its MD5 checksum should be `93ee1e55dbc19d70b170f7a1ab72d4a1` (`md5sum /lib/modules/ch341.ko`). Can you confirm those values?

stegemanGit, 2024-02-25:
> Try using a USB extension cord.

Great tip! It also worked in my case.

NobodyIT, 2024-11-04:
Hi all,

I will preface this by saying that I am not a Linux expert so I thank you for your patience and help.  
I have tried to install the usb driver on a DS223J but I always get an error at installation.

Some information about my system/test:

nname  Linux AMAC 5.10.55+ #72806 SMP Thu Sep 5 13:43:49 CST 2024 aarch64 GNU/Linux synology_rtd1619b_ds223j
File used:

i have followed all the instructions and wanted to install usb drivers for a usb zigbee devices on my synology. (No ttyusb or ttyacm device in /dev)

Unfortunately i am running dsm 7.2
(DSM 7.2-64570 Update 1)
(Linux MyCloud 4.4.302+ #64570 SMP Tue Jun 13 21:39:55 CST 2023 x86_64 GNU/Linux synology_geminilake_720+)

It is a 720+ geminilake synology. I did try these drives here: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2

None of those i could install with this command as described in the manual: insmod /lib/modules/cp210x.ko
I am always getting this error:

sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

NobodyIT, 2024-11-04:
Hi all,

I have followed tried to install usb drivers for a usb zigbee devices on my synology but i get some trouble.
The hardware is a synology DS223J - with DSM7.2 - (synology_rtd1619b_ds223j)

I downloaded the files from the following link;
https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1619b/dsm-7.2/modules/ch341.ko

when I try to install I get the following error:
sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

I am not a linux expert and almost certainly made some mistakes, any suggestions will be greatly appreciated.
Thanks in advance for any help.

robertklep, 2024-11-04:
@NobodyIT please follow the [download instructions](https://github.com/robertklep/dsm7-usb-serial-drivers/#downloading-a-module) *carefully*. You're not downloading from the correct URL.

NobodyIT, 2024-11-04:
@robertklep Thank you for now, I will try the procedure again

konclave, 2024-11-26:
@robertklep Thank you for your repo and the time you're spending supporting it. 

If I see the following messages in the `dmesg`, does that mean that drivers doesn't fit my system? 

```
cp210x: version magic '4.4.180+ SMP mod_unload ' should be '4.4.302+ SMP mod_unload '
[ 1156.402556] ch341: version magic '4.4.180+ SMP mod_unload ' should be '4.4.302+ SMP mod_unload '
```

That is the OS version: `Linux NoisyDemon 4.4.302+ #72806 SMP Thu Sep 5 13:44:44 CST 2024 x86_64 GNU/Linux synology_geminilake_720+`

And I've installed the drivers from this path: `/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/`

konclave, 2024-11-27:
@robertklep I have DSM 7.2.2-72806.

robertklep, 2024-11-29:
@konclave you can probably use `ttyACM` instead of `ttyUSB` without any issues. The difference is that `ttyACM` devices are created by the `cdc-acm.ko` driver and the `ttyUSB` devices are created by the drivers that depend on `usbserial` (which also means that you don't even need the drivers that I provide since `cdc-acm.ko`, which comes with DSM still, should be sufficient)

robertklep, 2024-11-29:
@d3nbo the _exact_ same problem? With an incorrect `version magic` error?

d3nbo, 2024-11-29:
Good news, it's runnig now. As I am not an IT expert, still a little clueless

I did not have the exact same, but ttyUSB0 was missing suddenly, when it was working before. I have a RS2418+, so I used denverton.

Problem was when using /usr/local/etc/rc.d/usb-serial-drivers.sh start

/usr/local/etc/rc.d/usb-serial-drivers.sh: line 7: syntax error near unexpected token `newline'
/usr/local/etc/rc.d/usb-serial-drivers.sh: line 7: `<!DOCTYPE html>'

Thank you for your great work!!

robertklep, 2024-11-29:
@d3nbo sounds like you may not have downloaded the script correctly.

Davidou2510, 2025-02-08:
@robertklep 

Hello, 
Have the same "invalid module format" with dsm 7.2.2-72806 update 3
DS224+ geminilake

Don't understand what I did wrong as I already used the same command lines on another syno (appololake this time) and it worked perfectly. 

Is the last DSM update the problem ? 

Bests

robertklep, 2025-02-08:
@Davidou2510 can you run the command `file` against any of the modules that you downloaded? 

So for example:
```
file /lib/modules/ch341.ko
```

Davidou2510, 2025-02-09:
@robertklep 
Thanks for your answer

The command 'file /lib/modules/ch341.ko' has returned as "command not found" in /lib/modules
Am I in the right folder ? Or should I run this command in cd /dev ? 

Now, after new dsm install, I can see the ttyUSB0. 
The new problem is that ZHA cannot connect to it. 

Also, the rndis_host.ko file returned a "unknown symbol in module". Is that the problem ?

Bests
David

Davidou2510, 2025-02-09:
@robertklep 

My bad. Made a mistake when creating Home Assistant and didn't include zha services...

Everything seems to run fine now.

Let you know. 

Thanks for your help

robertklep, 2025-03-03:
@Freedom1301 if you have the same problem, it's for the same reason: you didn't follow [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers/#downloading-a-module) correctly.

Freedom1301, 2025-03-03:
> [@Freedom1301](https://github.com/Freedom1301) if you have the same problem, it's for the same reason: you didn't follow [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers/#downloading-a-module) correctly.

I already read many times...and I used links bellow. It is correct?

Drivers links:
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/99b34617b33fe5e6a2768c63fb4bbc4a810c0386/modules/geminilake/dsm-7.2/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/99b34617b33fe5e6a2768c63fb4bbc4a810c0386/modules/geminilake/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/99b34617b33fe5e6a2768c63fb4bbc4a810c0386/modules/geminilake/dsm-7.2/pl2303.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/99b34617b33fe5e6a2768c63fb4bbc4a810c0386/modules/geminilake/dsm-7.2/rndis_host.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/99b34617b33fe5e6a2768c63fb4bbc4a810c0386/modules/geminilake/dsm-7.2/ti_usb_3410_5052.ko

USB serial start
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/refs/heads/main/usb-serial-drivers.sh

Sorry but whats wrong?

Thank you.

robertklep, 2025-03-03:
@Freedom1301 those are correct, assuming that your NAS uses the Gemini Lake platform. So what is the problem exactly?

robertklep, 2025-03-04:
@Freedom1301 does `/dev/ttyACM0` exist? If so, that's probably the one you need.

Freedom1301, 2025-03-04:
> [@Freedom1301](https://github.com/Freedom1301) does `/dev/ttyACM0` exist? If so, that's probably the one you need.
> 
> If it doesn't exist, please post the output of the following commands:
> 
> ```
> lsusb -cui
> ```
> 
> And:
> 
> ```
> lsmod | grep usbserial
> ```

/dev/ttyACM0 doesn´t exist.

lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0451:16ae:6165 00  2.00   12MBit/s 100mA 1IF  (Texas Instruments CC2531 USB Dongle ffffffd1ffffffbdffffffdaffffffb4)
  1-1:1.0         (IF) ff:ff:ff 1EP  (Vendor Specific) usbfs
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004430EB36DF32)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)

lsmod | grep usbserial
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  18 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

robertklep, 2025-03-04:
@Freedom1301 all the drivers have been loaded properly, so either your device isn't supported or there's a hardware issue (which for instance can happen if you don't have an extension cord).

Can you give a link to exactly which device you have, and which firmware it's running? FWIW, the CC2531 is an outdated device and I wouldn't really recommend using it as a coordinator.

Freedom1301, 2025-03-04:
@robertklep 
I am using extentsion cord.
Link https://www.ti.com/product/CC2531#description for my coordinator.

If you mean that is too old, coould you recomend some different type?

Thanks

Freedom1301, 2025-03-04:
@robertklep 
maybe this one ?
ZigBee EFR32MG21

Thanks

robertklep, 2025-03-04:
@Freedom1301 I wouldn't recommend anything with an EFR32, the CC2652's are much more stable.

I don't know which serial chip that CC2531 dongle uses, but clearly it's not supported by any of the drivers that I provide 😢

Freedom1301, 2025-03-04:
@robertklep so I need buy the new one coordinator. Thank you very much for you support. 👌

MarcelAarsman, 2025-03-06:
@robertklep my DS224+ is also using GeminiLake platform

The insmod of rndis_host.ko is showing an error. Is this a problem? The rest works fine.

insmod: ERROR: could not insert module /lib/modules/rndis_host.ko: Unknown symbol in module

robertklep, 2025-03-07:
@MarcelAarsman no, that's not a problem. You can just remove the file.

closed 2025-03-07 by robertklep

mihir-liverpool, 2025-05-18:
Hi Robert, appreciate all the help here. I am following your download and install instructions listed [here](https://github.com/robertklep/dsm7-usb-serial-drivers?tab=readme-ov-file#downloading-a-module) . I am using DS223J and I was able to download all the drivers from https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1619b/dsm-7.2 . I have succesfully moved them over to /lib/modules and downloaded and ran usb-serial-drivers.sh too. However I am still able to receive device assignment to my sonoff zigbee dongle. 

mihirshelar@DS223J:~$ lsusb -cui
 |__5-1.1.4 10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 ec70019e0274ef11a47de61e313510fd)
    5-1.1.4:1.0   (IF) ff:00:00 2EPs ()  

mihirshelar@DS223J:~$ lsmod | grep serial
usbserial              28672  1 ftdi_sio
usbcore               233472  10 ftdi_sio,usbserial,xhci_hcd,usblp,cdc_acm,usb_storage,rtk_usb_manager,xhci_pci,uas,xhci_plat_hcd

Can you help guide on what I could be missing?

mihir-liverpool, 2025-05-18:
When I cd /lib/modules I am able to see the ch341.ko, pl2303.ko  and cp210x.ko but I dont see them on lsmod | grep serial even though I have downloaded and I can view it in file path /usr/local/etc/rc.d$ 
I have then ran sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start . I am unable to figure out what I could be missing?

robertklep, 2025-05-18:
@mihir-liverpool try loading the modules manually:
```
sudo insmod /lib/modules/cp210x.ko
sudo insmod /lib/modules/ch341.ko 
```

mihir-liverpool, 2025-05-18:
mihirshelar@DS223J:/usr/local/etc/rc.d$ sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

From your guidance above it seems this is usually thrown when incorrect files are downloaded?

mihir-liverpool, 2025-05-18:
Thanks man, I figured it out - I was missing the 'raw' in the URL . 

mihirshelar@DS223J:/lib/modules$ lsmod | grep serial
usbserial              28672  4 pl2303,ftdi_sio,cp210x,ch341
usbcore               233472  13 pl2303,ftdi_sio,usbserial,xhci_hcd,cp210x,usblp,cdc_acm,usb_storage,rtk_usb_manager,xhci_pci,uas,ch341,xhci_plat_hcd

Finally got the device assignment 

      |__5-1.1.4 10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 ec70019e0274ef11a47de61e313510fd)
    5-1.1.4:1.0   (IF) ff:00:00 2EPs () cp210x ttyUSB0

mihir-liverpool, 2025-05-18:
mihirshelar@DS223J:~$ sudo chmod 666 /dev/ttyUSB0
Password: 
mihirshelar@DS223J:~$ docker run --device /dev/ttyUSB0 ...
docker: invalid reference format.
See 'docker run --help'.

Am I missing something on the final docker command?

robertklep, 2025-05-18:
@mihir-liverpool what's the full command line that you're using?

mihir-liverpool, 2025-05-18:
mihirshelar@DS223J:~$ docker ps
CONTAINER ID   IMAGE                                 COMMAND   CREATED        STATUS        PORTS     NAMES
f3bad3cb3193   homeassistant/home-assistant:latest   "/init"   3 months ago   Up 12 hours             homeassistant-home-synology

mihirshelar@DS223J:~$ docker run --device /dev/ttyUSB0 ...
docker: invalid reference format.
See 'docker run --help'.

mihirshelar@DS223J:~$  lsusb -cui
|__usb5          1d6b:0002:0510 09  2.00  480MBit/s 0mA 1IF  (Linux 5.10.55+ xhci-hcd xHCI Host Controller xhci-hcd.3.auto) hub
  |__5-1         2109:2817:9013 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              syno.ext.hub) hub
    |__5-1.1     05e3:0610:9224 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.0 Hub d6a3ebcbfbd5e7a5f6a3) hub
      |__5-1.1.4 10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 ec70019e0274ef11a47de61e313510fd)
    5-1.1.4:1.0   (IF) ff:00:00 2EPs () cp210x ttyUSB0 
|__usb6          1d6b:0003:0510 09  3.00 5000MBit/s 0mA 1IF  (Linux 5.10.55+ xhci-hcd xHCI Host Controller xhci-hcd.3.auto) hub
  |__6-1         2109:0817:9013 09  3.20 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.0 Hub              syno.ext.hub) hub

robertklep, 2025-05-19:
@mihir-liverpool are you _literally_ running `docker run --device /dev/ttyUSB0 ...`, with three dots at the end?

Why don't you use the built-in Docker UI that Synology provides?