robertklep/dsm7-usb-serial-drivers #112: module verification failed: signature and/or required key missing - tainting kernel

MesterJoda, 2023-12-11:
Hi @robertklep 
First of all, big thank you for your work. 
I'm loading the pl2303, loaded `usbserial.ko` first, but I'm getting this error in dmesg:
```
[  667.466419] usbcore: registered new interface driver usbserial
[  671.396572] pl2303: module verification failed: signature and/or required key missing - tainting kernel
[  671.407355] usbcore: registered new interface driver pl2303
[  671.413650] usbserial: USB Serial support registered for pl2303
[ 2154.836454] pl2303: version magic '4.4.180+ SMP mod_unload ' should be '4.4.302+ SMP mod_unload '
```

Running DS224+ (Linux NAS 4.4.302+ #69057 SMP Mon Nov 13 14:19:30 CST 2023 x86_64 GNU/Linux synology_geminilake_224+ )

I've tried to build the kernel sources myself, but I don't know where to clone your repo relative to the `/toolkit` folder. I've tried to copy `sources/4.4.x` directory to `/toolkit/source/4.4.x` but build fails with this error:
```
============================================================
                Start to run "Build Package"
------------------------------------------------------------
[geminilake] env PackageName=4.4.x /pkgscripts-ng/SynoBuild --geminilake -c --min-sdk 7.2 --parallel 4 4.4.x
Set cache size limit to 8.0 GB
Statistics zeroed
[INFO] projectList=4.4.x
[INFO] Start to build 4.4.x.
[SCRIPT] build script: //source/4.4.x/SynoBuildConf/build
[INFO] ======= Run build script =======
rm -rf *.o ch341.ko cp210x.ko rndis_host.ko pl2303.ko ti_usb_3410_5052.ko *.cmd
make -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules
make -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules
make -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make -C /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build M=//source/4.4.x modules
make[1]: Entering directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build'
  CC [M]  //source/4.4.x/ch341.o
  CC [M]  //source/4.4.x/ch341.o
  CC [M]  //source/4.4.x/ch341.o
  CC [M]  //source/4.4.x/ch341.o
fixdep: error opening depfile: //source/4.4.x/.ch341.o.d: No such file or directory
scripts/Makefile.build:286: recipe for target '//source/4.4.x/ch341.o' failed
make[2]: *** [//source/4.4.x/ch341.o] Error 2
make[2]: *** Deleting file '//source/4.4.x/ch341.o'
Makefile:1467: recipe for target '_module_//source/4.4.x' failed
make[1]: *** [_module_//source/4.4.x] Error 2
make[1]: Leaving directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.2/build'
Makefile:8: recipe for target 'pl2303.ko' failed
make: *** [pl2303.ko] Error 2
```
I'm willing to try to build the module myself, but I need a bit of guidance then.

MesterJoda, 2023-12-11:
Additional information. After trying a lot of different thing, I've managed to build the module by deleting all except pl2303 from Makefile, which looks like this:
```
kernel_mod = pl2303.ko

all: $(kernel_mod)

obj-m := pl2303.o

$(kernel_mod):
        make -C $(KSRC) M=$(PWD) modules

install: $(kernel_mod)
        mkdir -p $(DESTDIR)/4.4.x/ 2>/dev/null
        install $< $(DESTDIR)/4.4.x/

clean:
        rm -rf *.o $(kernel_mod) *.cmd
Makefile (END)
```
Funny thing is, the built `pl2303.ko` and one downloaded from here, the Github, are identical. What now?

robertklep, 2023-12-11:
The "tainted kernel" message is a warning, not an error, the module should be loaded regardless.

MesterJoda, 2023-12-11:
Hi @robertklep 
Than you for the fast response. Here is my output:
```
lsmod | grep pl2303
pl2303                  9022  0
usbserial              22278  1 pl2303
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,usbserial,ehci_hcd,ehci_pci,pl2303,usbhid,xhci_hcd,xhci_pci
```
Is this then correct behaviour? Ifthat is the case, I'll gladly close the issue :)

robertklep, 2023-12-11:
Looks like it loaded just fine 😅

MesterJoda, 2023-12-11:
How could I check the tty? I'm seeing only
```
cat /proc/tty/driver/usbserial
usbserinfo:1.0 driver:2.0
```
Shouldn't I see similar to the following output:
```
usbserinfo:1.0 driver:2.0
0: module:ch341 name:"ch341-uart" vendor:1a86 product:7523 num_ports:1 port:0 path:usb-0000:00:1d.0-1.2
```

robertklep, 2023-12-11:
No idea to be honest. Do you have a device connected with a PL2303 chip?

robertklep, 2023-12-11:
Do you think the 1-2 device is your USB-to-serial device? It doesn't show a name, only a serial number(?). Does it disappear when you unplug the device?

robertklep, 2023-12-11:
Very strange, since its USB device id (`1a86:55d3`) suggests it might be using a CH341 chip (or at least in the CH family).

You can try and see if the ch341.ko driver supports it, but I doubt it since I cannot find that specific device id in any of the drivers. It might simply be too new, and Synology's kernel sources too old 😬

MesterJoda, 2023-12-11:
Now that yu have mentioned it, I've loaded all of the modules, not just usbserial and pl2303 and I have a different output:
`lsusb -cui`
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         1a86:55d3:0445 02  1.10   12MBit/s 138mA 2IFs ( 5677003661)
  1-2:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
  1-2:1.1         (IF) 0a:00:00 2EPs () cdc_acm
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500350AB181C639)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)
```

robertklep, 2023-12-11:
It's there as `/dev/ttyACM0` now 😊

closed 2023-12-11 by MesterJoda