robertklep/dsm7-usb-serial-drivers #114: Epyc7002 drivers

xavier-thomas, 2023-12-22:
Hi,

Is there a possibility to add drivers for the Epyc7002 platform ?
I'm trying to use a zigbee USB device on a Synology SA6400

Kernel version is: 5.10

I can see the device being detected on the USB hub by the `lsusb` command, but doesn't seem to be mounted (I presume due to the lack of drivers).

[image]

robertklep, 2023-12-23:
Have you tried the drivers that come with DSM?
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/cdc-acm.ko
```

If that doesn't work: I'm in the process of migrating my Unraid server, which means I can't access the virtual machine in which I build these modules. So it may take some time before I will be able to build additional drivers.

xavier-thomas, 2023-12-27:
Hi @robertklep, that actually worked. I hadn't realised that DSM 7.2 shipped with native usb drivers.

closed 2023-12-27 by xavier-thomas

robertklep, 2023-12-27:
@xavier-thomas yeah they left _some_ drivers 😅