robertklep/dsm7-usb-serial-drivers #115: Devices not appearing on DS918+ with DSM 7.2.1

agambon, 2024-01-24:
I have been all through the closed issues here, and double-checked everything I could think of.
I have 2 devices plugged in: Z-wave stick, and google coral accelerator.
- The files are raw/binary, not blob/html.
- I downloaded the correct files (apollolake, 7.2)
- The Z-Wave stick is on a powered usb hub, and on an extender to eliminate interference.

lsmod | grep serial:
`usbserial              22534  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               210297  16 ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci`

lsusb -cui:
`|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         2109:2817:0783 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              000000000) hub
    |__1-1.3     10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Silicon Labs CP2102N USB to UART Bridge Controller dadea5c67a57ed11a9282ef21c62bc44)
   1-1.3:1.0      (IF) ff:00:00 2EPs () cp210x ttyUSB0
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00916F8F3A7D71)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host5 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1a6e:089a:0100 00  3.10 5000MBit/s 896mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad)
  2-1:1.0         (IF) fe:01:02 0EPs ()
  |__2-2         2109:0817:0783 09  3.20 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.0 Hub              000000001) hub`

I ran the provided usb-serial-drivers.sh file, and it seems to have executed without an error.
The cui dump above shows ttyUSB0, and this only displays when the device is plugged in - so that seems promising. But I still get nothing in DSM "External Devices" panel.

Help please!

robertklep, 2024-01-24:
Given that the device is available as `/dev/ttyUSB0`, it works as expected.

The External Devices panel is, AFAIK, only meant for USB storage devices, and neither of your devices are, so it makes sense they don't show up there.

What exactly are you trying to accomplish?

agambon, 2024-01-24:
Ah, ok. I misunderstood what that screen was for. I'm trying to get the devices passed through to docker (Home assistant). I am running the container as high privilege, so that should grant it all devices. But it is not detecting anything.

Koky05, 2024-01-24:
For allow HA work with these devices you need to add `device: /dev/ttyUSB0` into run command. For me also work bind it as volume `volume: /dev/ttyUSB0:/dev/ttyUSB0`. After that it was rocognized in HA.

agambon, 2024-01-24:
Ok, I thought that giving the container "high privilege" would allow it all devices. For some reason, when I give the docker container explicit access to that device, the container bootup hangs forever and doesn't come fully online.
However, that's not an issue with these drivers. Thanks all, I'll close this for now and keep troublshooting my container.

closed 2024-01-24 by agambon

agambon, 2024-01-24:
Update - I fixed a typo in my port declaration and got the container back online.
Still no indication that Home Assistant is receiving the USB device.

My docker, in case anyone knows:

`
version: "3.9"
services:
  homeassistant:
    container_name: homeassistant
    privileged: true
    restart: unless-stopped
    image: ghcr.io/home-assistant/home-assistant:stable
    devices:
      - /dev/ttyUSB0
    ports:
      - "8123:8123"
    volumes:
      - /volume1/docker/homeassistant:/config
      - /dev/ttyUSB0:/dev/ttyUSB0
    environment:
      TZ: "America/New_York"
`

agambon, 2024-01-24:
I tried expanded the `devices` section with a bunch of variants just to cover bases.
Still nothing.

```
    devices:
      - /dev/bus/usb:/dev/bus/usb # passes the USB Coral, needs to be modified for other versions
      - /dev/apex_0:/dev/apex_0 # passes a PCIe Coral, follow driver instructions here https://coral.ai/docs/m2/get-started/#2a-on-linux
      - /dev/dri/renderD128 # for intel hwaccel, needs to be updated for your hardware
      - /dev/ttyUSB0
      - /dev/ttyACM0
      - /dev
      - "/dev/ttyACM0:/dev/ttyACM0"
      - "/dev/ttyUSB0:/dev/ttyUSB0"
      - "/dev:/dev"
```

agambon, 2024-01-24:
UPDATE: Solved! Needed to run the Z-Wave JS UI docker container AND link that to the HA docker with a network bridge. Hopefully this helps someone in the future.