robertklep/dsm7-usb-serial-drivers #116: Epyc7002 modules for usb

error105, 2024-01-27:
Hi, there is any chance that u will prepare ch340 and other for Epyc7002 (SA6400) with kernel 5.10.55+ ?

robertklep, 2024-01-28:
I'll take a look. I assume that it's running DSM 7.2?

error105, 2024-01-28:
> I'll take a look. I assume that it's running DSM 7.2?
> 
> Also, what device are you looking to add support for? In my experience, the DSM-built-in `cdc-acm.ko` module already covers the most popular devices.

Yes, 7.2 DSM :) I have cc2652 zigbee coordinator with ch340 usb-uart, also I want connect my TevoTarantula to use it with Klipper on docker.

robertklep, 2024-01-28:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/epyc7002/dsm-7.2

(make sure to follow the [downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module))

error105, 2024-01-28:
> Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/epyc7002/dsm-7.2
> 
> (make sure to follow the [downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module))

Thanx, i will try evening and will let u know :)

error105, 2024-01-28:
> Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/epyc7002/dsm-7.2
> 
> (make sure to follow the [downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module))

Works perfect !! ur awesome.

robertklep, 2024-01-28:
Thanks for testing! 👍🏻

closed 2024-01-28 by robertklep