robertklep/dsm7-usb-serial-drivers #117: DSM 7.2.1 apollolake drivers

NoAdO, 2024-02-05:
**Type of device**
My device is SONOFF ZB Dongle-P based on cc2562 , my DSM system is 7.2.1
Linux NoAdO_DSM 4.4.302+ #69057 SMP Fri Jan 12 17:02:59 CST 2024 x86_64 GNU/Linux synology_apollolake_918+

**Have the modules loaded?**
Yes, i've used 
```wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/apollolake/dsm-7.2/ch341.ko'
sudo cp ch341.ko /lib/modules
sudo insmod /lib/modules/ch341.ko
```
ended with no errors. Start script was added earlier with ch341.ko put in /lib/modules for DSM 7.2 and it worked perfect.

**Output of lsusb**
```
/usr/syno/bin/lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  1-1:1.0         (IF) ff:00:00 2EPs ()
```
But now there is no ttyUSB attachment

Also, if i try to insmod there's no corresponding log at /var/log/messages. But if i try to rmmod and then insmod there is a message about removing but no message about installing:
```
2024-02-05T15:17:39+03:00 NoAdO_DSM kernel: [ 5156.272871] Module [ch341] is removed.
```

I suggested that smth changed from 7.2 to 7.2.1 and stared to follow build from source manual at ubuntu subsystem for windows. But there's no EnvDeploy for 7.2.1, it's listed if i ask, but there's no file for downloading.
```root@pc:/toolkit/pkgscripts-ng# ./EnvDeploy -v 7.2.1 --list
[2024-02-05 15:19:53,944] INFO: Available platforms: avoton braswell bromolow grantley alpine alpine4k monaco armada38x kvmcloud kvmx64 rtd1296 broadwellnk denverton apollolake armada37xx purley v1000 broadwell geminilake broadwellntbap r1000 broadwellnkv2 rtd1619b epyc7002
root@pc:/toolkit/pkgscripts-ng# ./EnvDeploy -v 7.2.1 -p apollolake
[2024-02-05 15:20:07,516] INFO: Download... https://dataupdate7.synology.com/toolchain/v1/get_download_list?identify=toolkit&version=7.2.1&platform=base
[2024-02-05 15:20:08,325] INFO: Download... https://dataupdate7.synology.com/toolchain/v1/get_download_list?identify=toolkit&version=7.2.1&platform=apollolake
[2024-02-05 15:20:08,985] ERROR: TarballNotFoundError
[2024-02-05 15:20:08,985] ERROR: Needed file not found! /toolkit/toolkit_tarballs/base_env-7.2.1.txz
[2024-02-05 15:20:08,986] ERROR: ./EnvDeploy -v 7.2.1 -p apollolake failed!```
I've found that 
```[2024-02-05 15:38:23,356] INFO: Download... https://dataupdate7.synology.com/toolchain/v1/get_download_list?identify=toolkit&version=7.2.1&platform=apollolake```
is successfull, the broblem is in missing /toolkit/toolkit_tarballs/base_env-7.2.1.txz file.

If i try to use USB hub nothing changes:
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-3         0bda:5411:0117 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
    |__1-3.1     10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
   1-3.1:1.0      (IF) ff:00:00 2EPs ()```

If it helps, looks like drivers loaded successfully:
```lsmod | grep serial
usbserial              22534  2 ch341,ftdi_sio
usbcore               210297  14 ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,usbserial,ehci_hcd,ehci_pci,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci```

So for now i am totally confused and do not know what to try next. Looks like 7.2 drivers for 7.2.1 are not working, i can't compile a newer version and my Smart home is not smart at all now :D

robertklep, 2024-02-05:
The CC2562 dongle has a CP2102 serial-to-USB chip on board, so you need the `cp210x.ko` driver.

NoAdO, 2024-02-05:
Thanks a lot and shame on me! I should try to get all files, mb ch341 is left for previous dongle, and i didn't check that. I'll investigate to VM installation of HAOS, a bit tired of this quest every time i update.

closed 2024-02-05 by NoAdO