robertklep/dsm7-usb-serial-drivers #118: Unable to view ttyACM0 or ttyUSB0 after running install scripts on DSM7.1 Synology for Sonoff Zigbee USB

charsiufan, 2024-02-07:
**Type of device**
Synology DS1815+ (Avoton), DSM: DSM 7.1.1-42962 Update 6
Sonoff zigbee 3.0 usb dongle plus - shows up as a 'CP210x UART Bridge - Silicon Labs' in the NAS info section

Used this to get the modules: 
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/avoton/dsm-7.1/cp210x.ko'
no error reported when executing # /usr/local/etc/rc.d/usb-serial-drivers.sh start
**Have the modules loaded?**
Are you sure that the modules you downloaded from this repository have loaded properly? To check, you can run the following commands (after downloading/installing the modules):
```
sudo insmod /lib/modules/usbserial.ko No error
sudo insmod /lib/modules/ftdi_sio.ko No error
sudo insmod /lib/modules/cdc-acm.ko No error
sudo insmod /lib/modules/cp210x.ko No error
sudo insmod /lib/modules/ch341.ko insmod: ERROR: could not load module /lib/modules/ch341.ko: No such file or directory
sudo insmod /lib/modules/pl2303.ko insmod: ERROR: could not load module /lib/modules/pl2303.ko: No such file or directory
```
These commands shouldn't return any errors. If they do, please post them here (unless the error is _"Invalid module format"_, in which case you should [follow the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) carefully).

**Output of lsusb**
Please include the output of the following command:
```
/usr/syno/bin/lsusb -cui
```

```
// PASTE OUTPUT HERE

myadmin@Nas:/dev$ /usr/syno/bin/lsusb -cui
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65005B616B6DAE21)
   1-1.1:1.0      (IF) 08:06:50 2EPs () usb-storage host10 (synoboot)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-2         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus b423b889db9ded11a05ed1a5a7696f5d)
  2-2:1.0         (IF) ff:00:00 2EPs () cp210x ttyUSB0
  |__2-4         051d:0002:0006 00  1.10  1.5MBit/s 2mA 1IF  (American Power Conversion Back-UPS CS 500 FW:808.q14 .I USB FW:q14  4B2027P23738  )
  2-4:1.0         (IF) 03:00:00 1EP  () usbfs
```

robertklep, 2024-02-07:
It's right there:
```
  |__2-2         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus b423b889db9ded11a05ed1a5a7696f5d)
  2-2:1.0         (IF) ff:00:00 2EPs () cp210x ttyUSB0
```

So it's accessible through `/dev/ttyUSB0`

charsiufan, 2024-02-07:
Thank you, my apologies. I reran the lsusb command right before I posted as it wasn't showing before and didn't reread the output. I guess it was chugging away in the background. Thanks for checking and for sharing this project!

closed 2024-02-07 by charsiufan