robertklep/dsm7-usb-serial-drivers #121: /dev/ttyUSB0 is not there anymore (nothing else changed)

lfilho, 2024-04-09:
**Type of device**
Sonoff Dongle
DSM 7.2
Linux <name> 4.4.302+ #69057 SMP Fri Jan 12 17:02:28 CST 2024 x86_64 GNU/Linux synology_geminilake_220+

**Have the modules loaded?**
Yes. Or I think they are. When i manually invoke `sudo insmod /lib/modules/usbserial.ko`, it returns: `insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists`. Should it have returned nothing at all?

**Output of lsusb**

```
lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus 1637575acb60ec11b230397625bfaa52)
  1-2:1.0         (IF) ff:00:00 2EPs () usbfs
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65001C106D345A24)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)
```

Note: it appears there isn't a device assigned as expected. It used to show `/dev/ttyUSB0`

---
Other info:
I didn't change the dongle nor the cable it is connected to synology. Everything was working until my zigbee devices stopped working last night, and while investigating why i eventually saw `/dev/ttyUSB0` wasn't there as it used to. I've been trying to get it back there to no avail.

lfilho, 2024-04-09:
If it helps, here's how I installed the modules:

```sh
#!/bin/bash

set -eo pipefail
echo "Make sure you runt his script as root or with sudo"

echo "Downloading modules"
cd /lib/modules || exit 1
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ch341.ko'
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko'
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/pl2303.ko'
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/rndis_host.ko'
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/ti_usb_3410_5052.ko'

echo "Downloading script to start usb modules"
cd /usr/local/etc/rc.d || exit 1
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/usb-serial-drivers.sh'
chmod +x /usr/local/etc/rc.d/usb-serial-drivers.sh

echo "Starting usb modules"
/usr/local/etc/rc.d/usb-serial-drivers.sh start

exit 0
```

robertklep, 2024-04-10:
Can you check which modules are loaded?
```
lsmod | grep usbserial
```

lfilho, 2024-04-11:
Uuugh, I just ran the command again and now it's there?!? I didn't change a thing, not even a restart! 🤯
I'll close this issue since i have no clue how to reproduce it and keep a closer eye if it happens again in the future. Sorry and thanks for the help!

lfilho, 2024-04-11:
Oh! Just happened again! So, after the restart it was there, and I was playing with zigbee2mqtt addon via its Addon UI, inside home assistant, inside a VM, inside synology.

Here's the output of `lsmod | grep usbserial`:
```
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  18 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci
```

The reason I'm messing with zigbee2mqtt in the first place was that it stopped working out of the blue, apparently a known issue to be fixed on the next version. Do you think it could be either z2m, HA or Synology crashing the usb module somehow? Any other ideas?

robertklep, 2024-04-11:
I have no idea, but I would suspect a hardware issue, either with your dongle or the cable.

When it happens, are any relevant messages shown in the output of `sudo dmesg`?

lfilho, 2024-04-11:
Oof, that's rough.

Yeah, here it, grepped for "usb":

```
sudo dmesg | grep -i usb
[    0.000000] ACPI: SSDT 0x0000000077ABE000 000271 (v02 INSYDE UsbCTabl 00000003 ACPI 00040000)
[   12.529189] ACPI: bus type USB registered
[   12.533710] usbcore: registered new interface driver usbfs
[   12.539859] usbcore: registered new interface driver hub
[   12.545816] usbcore: registered new interface driver ethub
[   12.551959] usbcore: registered new device driver usb
[   12.585526] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 1
[   12.804710] hub 1-0:1.0: USB hub found
[   13.052133] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 2
[   13.060495] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   13.069772] hub 2-0:1.0: USB hub found
[   13.336947] usb 1-2: new full-speed USB device number 2 using xhci_hcd
[   13.617302] usb 1-4: new high-speed USB device number 3 using xhci_hcd
[   32.157780] systemd[1]: Starting unbind USB device...
[   32.188373] systemd[1]: Created slice USBCopy's slice.
[   32.199503] systemd[1]: Starting USBCopy's slice.
[   32.386428] systemd[1]: Started unbind USB device.
[   32.545913] systemd[1]: Starting Mount usb fs...
[   35.080545] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   35.131267] uhci_hcd: USB Universal Host Controller Interface driver
[   37.015444] usbcore: registered new interface driver usblp
[   37.076011] usbcore: registered new interface driver usbhid
[   37.082240] usbhid: USB HID core driver
[   38.335058] usb-storage 1-4:1.0: USB Mass Storage device detected
[   38.342088] scsi host2: usb-storage 1-4:1.0
[   38.346863] usbcore: registered new interface driver usb-storage
[   38.461930] usbcore: registered new interface driver uas
[   75.552437] usbcore: registered new interface driver usbserial
[   75.577194] usbcore: registered new interface driver ftdi_sio
[   75.583640] usbserial: USB Serial support registered for FTDI USB Serial Device
[   75.636575] usbcore: registered new interface driver cdc_acm
[   75.642921] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   75.922087] usbcore: registered new interface driver cp210x
[   75.928332] usbserial: USB Serial support registered for cp210x
[   75.940737] usb 1-2: cp210x converter now attached to ttyUSB0
[   75.998792] usbcore: registered new interface driver ch341
[   76.004954] usbserial: USB Serial support registered for ch341-uart
[   76.046541] usbcore: registered new interface driver pl2303
[   76.052804] usbserial: USB Serial support registered for pl2303
[   76.095820] usbcore: registered new interface driver ti_usb_3410_5052
[   76.103083] usbserial: USB Serial support registered for TI USB 3410 1 port adapter
[   76.111687] usbserial: USB Serial support registered for TI USB 5052 2 port adapter
[82046.260558] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[82058.619624] usb 1-2: reset full-speed USB device number 2 using xhci_hcd
```

robertklep, 2024-04-11:
I think these two suggest that as far as the OS is concerned, the dongle has been disconnected physically:
```
[82046.260558] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[82058.619624] usb 1-2: reset full-speed USB device number 2 using xhci_hcd
```

So might be worthwhile to check if another cable (or even directly plugging it into the NAS, even though that's usually not a good idea due to interference issues) might solve it.

lfilho, 2024-04-11:
Hmmm but in that case, if it got physically disconnected, shouldn't it NOT
appear on the lsusb command's output?

robertklep, 2024-04-12:
Fair point, but it doesn't change the fact that this looks like a hardware issue. Besides that, I only compile the modules and have no idea how they actually work or how to solve issues with them 😅

mager33, 2024-04-21:
I'm stuck at the very same after installing DSM7.2.1-69057 Update 4 on my DS220+ (which was working totally fine with the Sonoff Zigbee 3.0 Stick. I followed all those steps and several variantions I found on the net, but no success: there is no /dev/ttyUSB0 any more.

robertklep, 2024-04-21:
@mager33 what is the output of `lsmod | grep usbserial` and `lsusb -cui`?

mager33, 2024-04-21:
`usbserial              22278  7 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  17 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci
`

`|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus 689ec39eeed8eb11950f0a15b9da5f8b)
  1-2:1.0         (IF) ff:00:00 2EPs () cp210x ttyUSB0
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology Diskstation 7F0016AC131A2E80)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host2 (synoboot)
`

robertklep, 2024-04-22:
@mager33 is the dongle connected through an extension lead? If not, it should be 😊

1uba, 2024-04-27:
@robertklep you often mentioning to use the extension lead. So I bought AXAGON HUE-SA7BP 7-Port USB 3.0 CHARGING hub. My goal is to connect Jablotron 100 and Home Assistant (running in Container on Synology 920+, version: DSM 7.2.1-69057 Update 5). I already tried and read many topics about the issue - unfortunately I can’t still see `ttyusb` or `ttyacm` device in /dev.  I have a same config as @lfilho above. Let me attach all the logs. I hope you can help me.

```
$ uname -u
synology_geminilake_920+
```

```
$ lsmod | grep usbserial
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  17 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci
```

```
$ lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F001B0C8413F221)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host6 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
```

```
$ sudo dmesg | grep -i usb
[    0.000000] ACPI: SSDT 0x0000000077ABE000 000271 (v02 INSYDE UsbCTabl 00000003 ACPI 00040000)
[   35.261407] ACPI: bus type USB registered
[   35.265928] usbcore: registered new interface driver usbfs
[   35.272076] usbcore: registered new interface driver hub
[   35.278033] usbcore: registered new interface driver ethub
[   35.284176] usbcore: registered new device driver usb
[   35.317855] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 1
[   35.537049] hub 1-0:1.0: USB hub found
[   35.783756] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 2
[   35.792083] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   35.801409] hub 2-0:1.0: USB hub found
[   36.046074] usb 1-4: new high-speed USB device number 2 using xhci_hcd
[   36.554025] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   36.700267] usb 1-1: not running at top speed
[   37.073970] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   37.220220] usb 1-1: not running at top speed
[   37.593898] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   37.740154] usb 1-1: not running at top speed
[   38.113824] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   38.260128] usb 1-1: not running at top speed
[   38.633753] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   38.779980] usb 1-1: not running at top speed
[   39.153676] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   39.299906] usb 1-1: not running at top speed
[   39.328923] hub 1-1:1.0: USB hub found
[   39.435338] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[   39.481626] drivers/usb/core/hub.c (2962) Same device found. Change serial to ffffffd6ffffffa3ffffffebffffffc0 
[   39.494808] hub 2-1:1.0: USB hub found
[   57.486863] systemd[1]: Created slice synokernel-usbserial's slice.
[   57.499978] systemd[1]: Starting synokernel-usbserial's slice.
[   57.743742] systemd[1]: Starting unbind USB device...
[   58.055552] systemd[1]: Started unbind USB device.
[   58.248362] systemd[1]: Starting Mount usb fs...
[   59.032973] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   59.072147] uhci_hcd: USB Universal Host Controller Interface driver
[   60.747803] usbcore: registered new interface driver usblp
[   60.785207] usbcore: registered new interface driver usbhid
[   60.792104] usbhid: USB HID core driver
[   61.094603] usb-storage 1-4:1.0: USB Mass Storage device detected
[   61.101592] scsi host6: usb-storage 1-4:1.0
[   61.106368] usbcore: registered new interface driver usb-storage
[   61.130423] usbcore: registered new interface driver uas
[   85.824746] usbcore: registered new interface driver usbserial
[   85.900987] usbcore: registered new interface driver ftdi_sio
[   85.907464] usbserial: USB Serial support registered for FTDI USB Serial Device
[   85.956607] usbcore: registered new interface driver cdc_acm
[   85.962940] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   85.995437] usbcore: registered new interface driver cp210x
[   86.001715] usbserial: USB Serial support registered for cp210x
[   86.017597] usbcore: registered new interface driver ch341
[   86.023789] usbserial: USB Serial support registered for ch341-uart
[   86.055567] usbcore: registered new interface driver pl2303
[   86.061844] usbserial: USB Serial support registered for pl2303
[   86.101050] usbcore: registered new interface driver ti_usb_3410_5052
[   86.101066] usbserial: USB Serial support registered for TI USB 3410 1 port adapter
[   86.101076] usbserial: USB Serial support registered for TI USB 5052 2 port adapter
```

It seems that all drivers are loaded properly, the only weird thing is:
`[   35.792083] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.`

robertklep, 2024-04-28:
@1uba I don't see any device connected in the output of `lsusb` (not even the hub, by the looks of it).

Even if there's a device connected that isn't supported by any of the drivers, it should still be visible.

1uba, 2024-04-28:
Thank you for your quick reply @robertklep
After some investigation I figured that due to the water damage there was a problem with Jablotron main unit - and the servis connector was damaged. After its replacement, I am finally getting some progress.  Unfortunately I can’t still see any `ttyusb or ttyacm device in /dev`.  Do you know where could be a problem?

```
sudo dmesg | grep -i usb
...
[ 5038.758809] usb 1-1.4.3: new full-speed USB device number 6 using xhci_hcd
[ 5038.892521] hid-generic 0003:16D6:0008.0002: hiddev0: USB HID v1.11 Device [Jablotron ltd JA-100 Flexi] on usb-0000:00:15.0-1.4.3/input0
[ 5038.906791] usb-storage 1-1.4.3:1.1: USB Mass Storage device detected
[ 5038.914249] scsi host9: usb-storage 1-1.4.3:1.1
[ 5039.934168] sd 9:0:0:0: [usb1] 204800 512-byte logical blocks: (105 MB/100 MiB)
[ 5039.934410] sd 9:0:0:0: [usb1] Write Protect is on
[ 5039.934412] sd 9:0:0:0: [usb1] Mode Sense: 03 00 80 00
[ 5039.934876] sd 9:0:0:0: [usb1] No Caching mode page found
[ 5039.934878] sd 9:0:0:0: [usb1] Assuming drive cache: write through
[ 5039.945859]  usb1: p1
[ 5039.949000] sd 9:0:0:0: [usb1] Attached SCSI removable disk
[ 5039.987084] sd 9:0:0:1: [usb2] 2301952 512-byte logical blocks: (1.18 GB/1.10 GiB)
[ 5039.987240] sd 9:0:0:1: [usb2] Write Protect is on
[ 5039.987242] sd 9:0:0:1: [usb2] Mode Sense: 03 00 80 00
[ 5039.987423] sd 9:0:0:1: [usb2] No Caching mode page found
[ 5039.987425] sd 9:0:0:1: [usb2] Assuming drive cache: write through
[ 5039.997957]  usb2: p1
[ 5039.999247] sd 9:0:0:1: [usb2] Attached SCSI removable disk
[ 5044.063699] FAT-fs (usb1p1): FAT: nls_disk load default table
[ 5044.073118] FAT-fs (usb1p1): FAT: nls_io load default table
[ 5050.167794] FAT-fs (usb2p1): FAT: nls_disk load default table
[ 5050.177976] FAT-fs (usb2p1): FAT: nls_io load default table
```

```
$ lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0bda:5411:0001 09  2.10  480MBit/s 0mA 1IF  (Generic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-1.4     0bda:5411:0001 09  2.10  480MBit/s 0mA 1IF  (Generic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffc1) hub
      |__1-1.4.3 16d6:0008:0107 00  2.00   12MBit/s 100mA 2IFs (Jablotron ltd JA-100 Flexi 14004036027151)
    1-1.4.3:1.0   (IF) 03:00:00 1EP  () usbhid 
    1-1.4.3:1.1   (IF) 08:06:50 2EPs () usb-storage host9 (usb1 usb2)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F001B0C8413F221)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host6 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:0411:0001 09  3.20 5000MBit/s 0mA 1IF  (Generic USB3.2 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
```

robertklep, 2024-04-28:
@1uba it looks like it's presenting as two separate devices: a USB HID device (so it may act as a keyboard or a mouse or something similar) and a USB storage device (the drivers I provide only support USB _serial_ devices).

Since you're looking to add support for it to Home Assistant, take a look [at this integration](https://github.com/plaksnor/HASS-JablotronSystem).

1uba, 2024-04-28:
- **USB Storage device** - it works as volume for the Jablotron Link .exe application - usually used for configuration.
- **USB HID device** - I am confused why in my case its only connected as:
`[ 5038.892521] hid-generic 0003:16D6:0008.0002: hiddev0: USB HID v1.11 Device [Jablotron ltd JA-100 Flexi] on usb-0000:00:15.0-1.4.3/input0` ----> But I see is that others have there the **hidraw** so it works as _serial_:
`[2313475.624360] hid-generic 0003:16D6:0008.0001: hiddev96, hidraw: USB HID v1.11 Device [Jablotron It JA-100 Flexi] on usb-xhci-hed.6.auto-1.2/input`

----

I was originally planning to start with [HACS Jablotron 100+ integration](https://github.com/kukulich/home-assistant-jablotron100). Both - even the [HASS-JablotronSystem integration](https://github.com/plaksnor/HASS-JablotronSystem) are supposed to be configured with the _serial port_. It’s mentioned in their documentatoin: _"Note: Because my serial cable presents as a HID device there format is /dev/hidraw[x], others that present as serial may be at /dev/ttyUSB0 or similar."_

robertklep, 2024-04-29:
`hidraw` isn't serial, it's a raw USB HID interface. Have you checked if you have a `/dev/hidraw*` device entry?

ElJaeger, 2024-10-31:
Facing the same issue after upgrading to DSM 7.2.2-72806 yesterday.

/lsmod | grep usbserial 
is giving me the following output:

/usbserial              22278  1 ftdi_sio
usbcore               209065  13 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,usbserial,ehci_hcd,ehci_pci,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

ElJaeger, 2024-10-31:
sudo insmod /lib/modules/cdc-acm.ko
insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: File exists

sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

sudo insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format

sudo insmod /lib/modules/pl2303.ko
insmod: ERROR: could not load module /lib/modules/pl2303.ko: No such file or directory

robertklep, 2024-10-31:
@ElJaeger are you sure you downloaded the files correctly? The error "Invalid module format" suggests otherwise.

ElJaeger, 2024-10-31:
I just repeated the steps in your .readme. After that, I once verified that all 4 files are located in /lib/modules. But still, I get this error.

I'm working on a Synology 220+ which should use Geminilake inside, so I downloaded the files like this:
```shell
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.1/cp210x.ko'
```

robertklep, 2024-10-31:
@ElJaeger since you're on DSM 7.2, you should try the 7.2 modules (you're downloading the 7.1 module now).

ElJaeger, 2024-11-14:
@robertklep Similar issue after latest DSM-Update (DSM 7.2.2-72806 Update 1) which should fix some security issues:

sudo insmod /lib/modules/cdc-acm.ko
insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: File exists

sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Unknown symbol in module

sudo insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Unknown symbol in module

sudo insmod /lib/modules/pl2303.ko
insmod: ERROR: could not insert module /lib/modules/pl2303.ko: Unknown symbol in module

`sudo dmesg`

gives me the following

`[ 1422.128535] cp210x: module verification failed: signature and/or required key missing - tainting kernel
[ 1422.139177] cp210x: Unknown symbol usb_serial_generic_open (err 0)
[ 1422.146124] cp210x: Unknown symbol usb_serial_generic_close (err 0)
[ 1422.153192] cp210x: Unknown symbol usb_serial_deregister_drivers (err 0)
[ 1422.160739] cp210x: Unknown symbol usb_serial_register_drivers (err 0)
[ 1440.773443] ch341: Unknown symbol usb_serial_generic_open (err 0)
[ 1440.780303] ch341: Unknown symbol usb_serial_generic_close (err 0)
[ 1440.787272] ch341: Unknown symbol usb_serial_handle_dcd_change (err 0)
[ 1440.794608] ch341: Unknown symbol usb_serial_generic_resume (err 0)
[ 1440.801626] ch341: Unknown symbol usb_serial_deregister_drivers (err 0)
[ 1440.809102] ch341: Unknown symbol usb_serial_generic_tiocmiwait (err 0)
[ 1440.816533] ch341: Unknown symbol usb_serial_register_drivers (err 0)
[ 1452.429636] pl2303: Unknown symbol usb_serial_handle_sysrq_char (err 0)
[ 1452.437132] pl2303: Unknown symbol usb_serial_generic_open (err 0)
[ 1452.444094] pl2303: Unknown symbol usb_serial_generic_close (err 0)
[ 1452.451161] pl2303: Unknown symbol usb_serial_handle_dcd_change (err 0)
[ 1452.458603] pl2303: Unknown symbol usb_serial_deregister_drivers (err 0)
[ 1452.466140] pl2303: Unknown symbol usb_serial_handle_break (err 0)
[ 1452.473100] pl2303: Unknown symbol usb_serial_generic_tiocmiwait (err 0)
[ 1452.480633] pl2303: Unknown symbol usb_serial_register_drivers (err 0)`

Do you know if this is an issue on my system or are there some missing dependencies in your drivers?

excalibur25, 2024-11-14:
same error here on 620slim after DSM 7.2.2-72806 Update 1!
Do we need new modules for this version?

robertklep, 2024-11-15:
I think these errors mean the `usbserial.ko` module wasn't loaded.

robertklep, 2024-11-15:
@ElJaeger load it: 

```
sudo insmod /lib/modules/usbserial.ko
```

Followed by the other modules you require.

ElJaeger, 2024-11-15:
Thank you. Works again! :)

Arie6414, 2024-11-15:
I have something similar as well after an update. I build a script (with my two left hands). I do not have that much expertise on this one, but, maybe you guys can look at it. What I want to do after a reboot (most of the time it is after an update), I would like to ins mod the serial drivers automatically. 

`modprobe usbserial
modprobe ftdi_sio
modprobe cdc-acm
cd /lib/modules/
wget 'https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/geminilake/dsm-7.2/cp210x.ko'
insmod /lib/modules/cp210x.ko`

Do you have any. ideas how to make this more future proof?

excalibur25, 2024-11-15:
> @ElJaeger load it:
> 
> ```
> sudo insmod /lib/modules/usbserial.ko
> ```
> 
> Followed by the other modules you require.

for me it is not working when trying to load the module.

insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists

robertklep, 2024-11-15:
@Arie6414 that's the wrong URL for downloading the modules

robertklep, 2024-11-15:
@excalibur25 it's loaded but you're still getting "Unknown symbol usb_serial_*" errors?

Arie6414, 2024-11-15:
> @Arie6414 that's the wrong URL for downloading the modules.
> 
> Also, you can try [the instructions provided in the README](https://github.com/robertklep/dsm7-usb-serial-drivers/#installation) on how to load the modules at boot time.

oh ah, explains a lot. 
I will try to make a script that make sense on Synology. Thanks!

robertklep, 2024-11-16:
@Arie6414 "blob" is the HTML page that offers to download the (raw) `.ko` file. So instead of downloading the module, it downloads the downloadpage for the module 😄

Arie6414, 2024-11-16:
Aah that explains a lot😅 thnx… still in the learning mode
Met vriendelijke groet,

Arjen van de Belt
Verstuurd vanaf mijn IPhone

h9seb, 2024-11-17:
Hi robert & team

i tried my best to follow the most recent procedure, have ingested the usb drivers for a syno918+ (appollolake) and the script to start, all good BUT still no /dev/ttyUSB not ACM available.
What i'm trying to achieve is to put an USB WIFI dongle to provide access to my nas file service on the sino instead of using the ethernet interface, nothing else (no docker nor virtual machine stuff). reason to do that is that wifi perform much better than CPL/ETHERNET here.
 
heres what i have with the lsusb -cui cmd
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  **|__1-3         0bda:b812:0210 00  2.10  480MBit/s 500mA 1IF  (Realtek 802.11ac NIC 123456)
  1-3:1.0         (IF) ff:ff:ff 5EPs (Vendor Specific)**
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0098E9A4E5F118)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host5 (synoboot)

3 options
1- i'm totally offroad and what i'm trying to achieve (accessing the nad file service of the syno from a wifi usb dongle is not possible
2- the wifi dongle i'm using (Realtek 8812BU Wireless LAN 802.11ac USB NIC) isn't supported
3- the problem is between the chair and the keyboard (the most possible one).

any idea/comment/clue is really welcomed :) thanks

robertklep, 2024-11-17:
@h9seb the drivers in this repository aren't meant for anything else but serial-to-USB dongles, so WiFi dongles aren't supported at all.

I don't know whether DSM will support them out-of-the-box, but if it does, you should look at Control Panel > Network > Network Interface.

robertklep, 2024-11-18:
@h9seb DSM runs Linux so with the right drivers it _should_ work. Question is: are those drivers available? I don't know 🙁

An alternative could be to buy a cheap WiFi access point that can act as a repeater in your network. Such access points typically have an Ethernet port, so you would connect the NAS over Ethernet to the repeater (short cable), then add the repeater to your wireless network.

h9seb, 2024-11-18:
sounds like a plan! will test if possible to add this driver first then, after failing as i will, i wil go to the WIFI ap meshed!
really short on knowledge on ingesting a driver on the linux os of the syno. you know the distrib used? (at least the closest!)
if you have any procedure on easily do that, feel free, i ll test it!
thanks again

robertklep, 2024-11-18:
@h9seb sorry, I have no idea if there even _are_ WiFi driver packages available for DSM 🙁

h9seb, 2024-11-18:
it used to be supported in DSM 6.2 : https://www.synology.com/fr-fr/compatibility?search_by=category&category=usb_wifi_dongles&p=2
now its gone, such a shame!

h9seb, 2024-11-18:
probably a good reason why they did this, but its a shame

robertklep, 2024-11-18:
@h9seb with DSM 7, Synology removed a lot of USB device support (one of the reasons why I started this repository).

robertklep, 2024-11-18:
@h9seb sorry, won't be me, I moved away from Synology some time ago already. I still have one that serves as a simple storage server, but everything else is now running on a Ubuntu server.

maol01, 2025-01-29:
Probably not, do I need to remove files to run through a re-install?

robertklep, 2025-01-29:
@maol01 it looks like the modules are working just fine as they have been loaded. Perhaps your Z-Wave dongle was assigned a different device entry (like `/dev/ttyUSB1`). What's the output of `lsusb -cui`?

mangan01, 2025-01-29:
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (dresden elektronik ingenieurtechnik GmbH ConBee II DE2687318)
  1-1:1.0         (IF) 02:02:01 1EP  () usbfs 
  1-1:1.1         (IF) 0a:00:00 2EPs () usbfs 
  |__1-3         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Silicon Labs CP2102N USB to UART Bridge Controller 8a12e22c017ded11806ad9f3fdf7b791)
  1-3:1.0         (IF) ff:00:00 2EPs () usbfs 
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F009AE0DFE5F108)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host5 (synoboot)

robertklep, 2025-01-29:
@mangan01 just to be clear, you're also @maol01? Have you tried rebooting your NAS?

What's the output of `sudo dmesg | grep -i usb`?

mangan01, 2025-01-29:
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=19A0PDN996406 macs=001132be433b,001132be433c
[    0.000000] ACPI: SSDT 0x000000007AFE4000 00024A (v02 INSYDE UsbCTabl 00000003 ACPI 00040000)
[    0.000000] Kernel command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=19A0PDN996406 macs=001132be433b,001132be433c
[    0.000000] USB Vbus GPIO Control:
[    0.000000]  Gpp#13 - Host: dev_name:usb2        - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#13 - Host: dev_name:usb1        - Port:3 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#11 - Host: dev_name:usb2        - Port:2 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#11 - Host: dev_name:usb1        - Port:1 - Polarity: ACTIVE_HIGH
[   47.613150] ACPI: bus type USB registered
[   47.617691] usbcore: registered new interface driver usbfs
[   47.623867] usbcore: registered new interface driver hub
[   47.629835] usbcore: registered new interface driver ethub
[   47.636007] usbcore: registered new device driver usb
[   47.686720] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 1
[   47.906064] hub 1-0:1.0: USB hub found
[   47.938216] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 2
[   47.946545] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   47.955942] hub 2-0:1.0: USB hub found
[   48.175003] usb 1-1: new full-speed USB device number 2 using xhci_hcd
[   48.441975] usb 1-3: new full-speed USB device number 3 using xhci_hcd
[   48.701964] usb 1-4: new high-speed USB device number 4 using xhci_hcd
[   65.101230] systemd[1]: Created slice synokernel-usbserial's slice.
[   65.114362] systemd[1]: Starting synokernel-usbserial's slice.
[   65.140047] systemd[1]: Starting unbind USB device...
[   65.575954] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   65.601513] uhci_hcd: USB Universal Host Controller Interface driver
[   66.543300] systemd[1]: Started unbind USB device.
[   66.779645] systemd[1]: Starting Mount usb fs...
[   66.800260] usbcore: registered new interface driver usblp
[   66.853628] usbcore: registered new interface driver usbhid
[   66.864478] usbhid: USB HID core driver
[   67.366967] usb-storage 1-4:1.0: USB Mass Storage device detected
[   67.374005] scsi host5: usb-storage 1-4:1.0
[   67.378859] usbcore: registered new interface driver usb-storage
[  102.616104] usbcore: registered new interface driver usbserial
[  102.634130] usbcore: registered new interface driver ftdi_sio
[  102.640698] usbserial: USB Serial support registered for FTDI USB Serial Device
[  102.665692] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  102.671886] usbcore: registered new interface driver cdc_acm
[  102.678233] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  102.724955] usbcore: registered new interface driver cp210x
[  102.731224] usbserial: USB Serial support registered for cp210x
[  102.748528] usb 1-3: cp210x converter now attached to ttyUSB0
[  102.773525] usbcore: registered new interface driver ch341
[  102.779718] usbserial: USB Serial support registered for ch341-uart
[  102.816692] usbcore: registered new interface driver pl2303
[  102.822987] usbserial: USB Serial support registered for pl2303
[  102.833857] usbcore: registered new interface driver ti_usb_3410_5052
[  102.841155] usbserial: USB Serial support registered for TI USB 3410 1 port adapter
[  102.849801] usbserial: USB Serial support registered for TI USB 5052 2 port adapter
[  247.529242] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  263.137591] usb 1-3: reset full-speed USB device number 3 using xhci_hcd
[ 2345.368762] usb 1-3: USB disconnect, device number 3
[ 2359.622470] usb 1-3: new full-speed USB device number 5 using xhci_hcd
[ 2359.772395] usb 1-3: cp210x converter now attached to ttyUSB0
[ 2629.872009] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 2708.519749] usb 1-3: USB disconnect, device number 5
[ 2721.993997] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[ 2722.145117] usb 1-3: cp210x converter now attached to ttyUSB0
[ 2752.846171] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 6539.850728] usb 1-3: USB disconnect, device number 6
[ 6544.378660] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[ 6544.527996] usb 1-3: cp210x converter now attached to ttyUSB0
[ 6565.293873] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0

The letters USB all come up red in the text

robertklep, 2025-01-29:
@mangan01 your devices keep getting disconnected, that's a hardware issue. Perhaps a faulty extension cable? Or, if you're not using an extension cable, try using one.

mangan01, 2025-01-29:
I lost patience moved over to a MAc-mini with Ubuntu, now everything is up again. Even if Ubuntu is unsupported for Supervised version of HA

closed 2025-01-29 by robertklep