robertklep/dsm7-usb-serial-drivers #125: XPEnology DSM 7.2 update 5 (No

pepegragera, 2024-06-25:
**Type of device**
Proliant Microserver N36L with XPEnology DS3622+ DSM 7.2.1 Update 5
Zgbee USB 3.0

The drivers used are these:
wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/ch341.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/cp210x.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/pl2303.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/rndis_host.ko
wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/broadwellnk/dsm-7.2/ti_usb_3410_5052.ko

And log is....
root@Proliant:/lib/modules# cd /lib/modules
root@Proliant:/lib/modules# cd /usr/local/etc/rc.d
root@Proliant:/usr/local/etc/rc.d# ./usb-serial-drivers.sh start
root@Proliant:/usr/local/etc/rc.d# lsusb -cui
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:13.2) hub
  |__2-2         0624:0248:0000 00  2.00  480MBit/s 2mA 3IFs (Avocent USB Composite Device-0 20090730)
  2-2:1.0         (IF) 03:01:01 1EP  () usbhid 
  2-2:1.1         (IF) 03:01:02 1EP  () usbhid 
  2-2:1.2         (IF) 03:01:02 1EP  () usbhid 
  |__2-3         f400:f400:0200 00  2.00  480MBit/s 100mA 1IF  (USB Disk 2.0 0978981064953095859)
  2-3:1.0         (IF) 08:06:50 2EPs () usb-storage host4 (synoboot)
root@Proliant:/usr/local/etc/rc.d# lsmod | grep usbserial
usbserial              22278  2 ch341,ftdi_sio
usbcore               208682  14 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,usbserial,ehci_hcd,ehci_pci,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

robertklep, 2024-06-25:
I don't see a serial device being attached (unless it's the "Avocent USB Composite Device", but that's a USB HID device, not a serial device)?

pepegragera, 2024-06-25:
Of course, the problem is precisely that the device does not detect

robertklep, 2024-06-25:
That's a hardware issue though, a connected USB device should appear in the output of `lsusb` regardless of whether a proper driver is available or not.

pepegragera, 2024-06-25:
I think it must be something with the Xpenology charger (emulator), any USB I connect is not detected by the DSM, I will continue looking for a solution for this problem.

robertklep, 2024-06-26:
Sounds like something I can't help with, so I'll close the issue for now 😊

closed 2024-06-26 by robertklep