robertklep/dsm7-usb-serial-drivers #126: My system is DSM is missing the cdc-acm.ko

pacolvin, 2024-07-29:
I posed this under another issue and now in this new issue.

My system is DSM is missing the cdc-acm. Anyone know where I may be able to get it? I can't get my Sonoff Zigbee 3 dongle to work after downloading and starting all of the other drivers. Not sure but maybe missing the cdc-acm.ko could be the reason.

From @robertklep 
Can you post the output of these commands:

lsmod | grep serial
lsusb -cui

Here are the results of those commands:

~$ lsmod | grep serial
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio

~$ lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:09:00.3) hub
  |__1-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus 9c36c28d24a6ed1190b7e8a32981d5c7)
  1-1:1.0         (IF) ff:00:00 2EPs () usbfs 
  |__1-3         0764:0601:0200 00  1.10  1.5MBit/s 50mA 1IF  (CPS UPS OR500 ffffff84ffffffb4ffffff81ffffffd3)
  1-3:1.0         (IF) 03:00:00 2EPs () usbfs 
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:09:00.4) hub
  |__3-2         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650037BEA681C648)
  3-2:1.0         (IF) 08:06:50 2EPs () usb-storage host10 (synoboot)

robertklep, 2024-07-29:
Yes, the missing extension seems to be the reason.

What's the output of this command?
```
uname -a
```

pacolvin, 2024-07-30:
Sorry.  I tried to reply via email but it did not appear here.

Here are the results.

~$ uname -a
Linux Colvin-NAS 4.4.302+ #69057 SMP Fri Jan 12 17:02:27 CST 2024 x86_64 GNU/Linux synology_v1000_1821+

robertklep, 2024-07-30:
I have an apollolake system myself which runs the same kernel version but I doubt that the module from my system will work on yours. I've uploaded it here: https://filebin.net/pzp661yvutqmt7x0

rcork, 2024-08-06:
Same problem. After a recent update, cdc-adm.ko is no longer in /lib/modules. Anyone have a guide on how to build the synology 4.4.x kernel and the cdc-acm module for v1000?

th0ma7, 2024-08-06:
There is a SynoCommunity package with serial port drivers if that helps? https://synocommunity.com/package/synokernel-usbserial 
You can certainly ask to add it to the x64 kernel build by opening an issue.

Actually, just checked (and rebuilt to confirm), it is included:
```
$ tree work-v1000-7.2/staging/
work-v1000-7.2/staging/
├── etc
│   └── synokernel-usbserial.cfg
├── lib
│   └── modules
│       └── v1000-7.2
│           └── 4.4.302+
│               └── drivers
│                   └── usb
│                       ├── class
│                       │   └── cdc-acm.ko
│                       └── serial
│                           ├── ch341.ko
│                           ├── cp210x.ko
│                           ├── ftdi_sio.ko
│                           ├── pl2303.ko
│                           ├── ti_usb_3410_5052.ko
│                           └── usbserial.ko
└── rules.d
    └── 60-synokernel-usbserial.rules
```
You'll have to copy the `*.ko` module to your `/lib/modules` directory after install.  Hopefully that a) works and b) helps.

rcork, 2024-08-06:
That SynoCommunity package worked and now i have ttyACM0 back. Thank you!

Cupid218, 2025-03-18:
Am I missing something?  The SynoCommunity SPK file doesn't contain cdc-acm.ko which is what I am missing and the issue is about.  How do I get it into /lib/modules

robertklep, 2025-03-18:
@Cupid218 which platform?

Cupid218, 2025-03-18:
RS1221 running 7.2.  v1000

robertklep, 2025-03-18:
Okay, so the v1000 platform.

The [SPK](https://packages.synocommunity.com/synokernel-usbserial/9/synokernel-usbserial.v9.f63134%5Bapollolake-avoton-braswell-broadwell-broadwellnk-broadwellnkv2-broadwellntbap-bromolow-denverton-geminilake-grantley-kvmx64-purley-r1000-v1000%5D.spk) contains the `cdc-acm.ko` file (inside `package.tgz`):
```
-rw-r--r--  0 root   root   366688 Jun 16  2024 ./lib/modules/v1000-7.2/4.4.302+/drivers/usb/class/cdc-acm.ko
```

Cupid218, 2025-03-18:
Thank You.  I'm a novice and don't know how this works.  I imported the SPK using "manual install" (I read somewhere that's how you do it) and it did not appear in the modules directory and neither does the v1000.  What's the process to get it there?

robertklep, 2025-03-18:
I never used SPK's so no idea how it works, but you can extract the file manually from a terminal (via ssh):
```
cd /tmp
wget 'https://packages.synocommunity.com/synokernel-usbserial/9/synokernel-usbserial.v9.f63134%5Bapollolake-avoton-braswell-broadwell-broadwellnk-broadwellnkv2-broadwellntbap-bromolow-denverton-geminilake-grantley-kvmx64-purley-r1000-v1000%5D.spk'
tar xvf *.spk
tar xvf package.tgz
sudo mv ./lib/modules/v1000-7.2/4.4.302+/drivers/usb/class/cdc-acm.ko /lib/modules
```

robertklep, 2025-03-19:
That should probably be `modprobe cdc_acm` (underscore instead of hyphen). There's no difference using `insmod` instead.

Cupid218, 2025-03-21:
I don't think things got loaded correctly.  even though ttyUSB0 exists nothings working.  I am at a loss on what is happing.  

I manually did the insmod of all the files in the usb-serial-drivers.sh.  For every file other than cdc-acm.ko I get the message "File exists", which I assume is a good thing.  For cdc-acm.ko I get "Device or resource busy".  Again a modprobe just says "not found"

I used sudo on everything this time to assure the owner is root.

If you have any suggestions it would be appreciated...

rcork, 2025-03-21:
@Cupid218 How did you install the SynoCommunity package? Did you go to Package Center, click Settings, add SynoCommunity location https://packages.synocommunity.com/. Then under Community look for the SynoKernel USB Serial Drivers and install it.

robertklep, 2025-03-21:
@Cupid218 if `ttyUSB0` exists, the drivers loaded properly and you'll have to elaborate what "nothing's working" means exactly.

Since the driver loaded, the issue is probably with the software you're trying to use (a setup/configuration issue).

Cupid218, 2025-03-21:
I installed the package via the instruction in another article by @robertklep https://github.com/robertklep/dsm7-usb-serial-drivers?tab=readme-ov-file.   (basically coping the .ko files and running usb-serial-drivers.sh to install (insmod) on each one)

The setup is to get openhab to recognize the drivers.  I didn't feel that I should put to much of that here as this isn't an openhab community site:  I mostly used these instructions:  https://community.openhab.org/t/synology-dsm-7-0-zwave-usb/124068

cdc-acm.ko appears to be a critical part of this driver and since it is not being loaded correctly I was centering on that as being the issue.

I had previously installed the two package 'SynoCli Kernel Tools' AND 'synoKernel USB Serial drivers' via Package Center.  No affect.

nutttz, 2025-07-21:
@Cupid218 did you ever get the cdc-acm.ko module to load.

I am having similar issues with the modprobe: FATAL: Module cdc-acm not found. message, even though the file exists in the module driectory.

nutttz, 2025-07-21:
That didn't work. I think it may have to do with the architecture of the DS925+ being v1000nk. I sourced the v1000 files from the synocommunity website as I couldn't find anything else anywhere with the v1000nk.

The output was:

insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: Invalid module format

nutttz, 2025-07-21:
the cdc-acm module was missing in the 925+ machine - but yes, I downloaded the module from the synocommunity website in the hope that it would work.

nutttz, 2025-07-22:
lsmod shows the usbserial and ftdi_sio modules loaded, but not the cdc-acm 

It is an interesting error. I don't know why the device would be busy.

nutttz, 2025-07-22:
and now the modprobe cdc-acm command gives the fatal erro that the Module is not found, despite the cdc-acm.ko bing present in the /lib/modules directory.

It is the same issue @Cupid218 had.

nvungoc, 2025-07-24:
I'm experiencing the same issue as @Cupid218 and @rcork. I've tried different things suggested above, but unfortunately no success. Here are the outputs: 

- lsmod | grep serial
usbserial              22278  1 ftdi_sio
-  lsusb -cui
|__usb1          ......
|__usb2         ......
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0a:00.4) hub
  |__3-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 d88825e08fc2ef1184a8c6138148b910)
  3-1:1.0         (IF) ff:00:00 2EPs () usbfs
  |__3-2        ......
- grep -ao 'vermagic=.*' /lib/modules/cdc-acm.ko
vermagic=4.4.302+ SMP mod_unload  GNU
- insmod /lib/modules/cdc-acm.ko
insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: Device or resource busy
- dmesg | tail
[  248.619434] audit: type=1325 audit(1753320736.947:1337): table=filter family=2 entries=51
[  248.702445] audit: type=1325 audit(1753320737.031:1338): table=filter family=2 entries=51
[  248.760596] audit: type=1325 audit(1753320737.089:1339): table=filter family=10 entries=66
[  248.774398] audit: type=1325 audit(1753320737.103:1340): table=filter family=10 entries=66
[  248.788421] audit: type=1325 audit(1753320737.117:1341): table=filter family=10 entries=66
[  248.801704] audit: type=1325 audit(1753320737.131:1342): table=filter family=2 entries=55
[  248.815393] audit: type=1325 audit(1753320737.144:1343): table=filter family=10 entries=70
[  248.828987] audit: type=1325 audit(1753320737.158:1344): table=filter family=2 entries=40
[  248.842305] audit: type=1325 audit(1753320737.171:1345): table=filter family=10 entries=39
[  256.461781] docker0: port 3(docker5054ba6) entered forwarding state

nvungoc, 2025-07-24:
> [@nvungoc](https://github.com/nvungoc) did you try adding SynoCommunity to the Synology Package Manager and then install the SynoKernel USB Serial Drivers? That's how i got it working on my DS1821+. No need to manual install SPKs, extract files, insmod, etc.

I got SynoCli Kernel Tools and SynoKernel USB Serial drivers installed. You mean there no need to install dsm7-usb-serial-drivers manually anymore? Btw, what dsm is your ds1821+ nas running on?

robertklep, 2025-07-24:
This seems strange: 
```
|__usb1          ......
|__usb2         ......
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0a:00.4) hub
  |__3-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 d88825e08fc2ef1184a8c6138148b910)
  3-1:1.0         (IF) ff:00:00 2EPs () usbfs
  |__3-2        ......
```
I'm not intimately familiar with the output of `lsusb`, but for me this suggests that there's a driver loaded that's assuming the Sonoff dongle is a USB mass storage device, which obviously isn't correct, but it could explain the _"Device or resource busy"_ error message.

It seems like mixing the SynoKernel modules with the ones I provide is murkying the waters.

nvungoc, 2025-07-24:
I've just uninstalled SynoKernel modules packages with the option "Erase data", but the output of lsusb -cui seems remain the same as above. What should I do to clean everything and start over from scratch? Thanks!

- lsusb -cui:
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0a:00.3) hub
  |__1-3         0764:0501:0001 00  1.10  1.5MBit/s 50mA 1IF  (CPS Value600ELCD fffffff0ffffffb3ffffffffffffffba)
  1-3:1.0         (IF) 03:00:00 1EP  () usbfs
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0a:00.3) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)
  2-1:1.0         (IF) ff:ff:00 3EPs () r8152 net/eth4
  |__2-2         0bc2:ab24:0100 00  3.00 5000MBit/s 144mA 1IF  (Seagate BUP Slim SL NA7YCJ4M)
  2-2:1.0         (IF) 08:06:50 2EPs () usb-storage host10 (usb1)
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0a:00.4) hub
  |__3-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 d88825e08fc2ef1184a8c6138148b910)
  3-1:1.0         (IF) ff:00:00 2EPs ()
  |__3-2         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650028CD2EEF6011)
  3-2:1.0         (IF) 08:06:50 2EPs () usb-storage host11 (synoboot)

nvungoc, 2025-07-24:
> [@nvungoc](https://github.com/nvungoc) I assume you rebooted after removing the packages?
> 
> Also, have you tried using the `cp210x.ko` driver for your Sonoff? I don't think it even needs `cdc-acm.ko` to work.

Yes, Sir, a reboot followed immediately after the removal of the packages. I wasn't able to insert module cp210x.ko, see the outputs below: 

- sudo -i 
- cd /lib/modules/
- wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/v1000/dsm-7.2/cp210x.ko
--2025-07-24 13:40:09--  https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/v1000/dsm-7.2/cp210x.ko
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'cp210x.ko'

cp210x.ko                                                       [ <=>                                                                                                                                     ] 179.01K  --.-KB/s    in 0.1s

2025-07-24 13:40:10 (1.38 MB/s) - 'cp210x.ko' saved [183304]
- insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

robertklep, 2025-07-24:
@nvungoc please follow [the downloading instructions](https://github.com/robertklep/dsm7-usb-serial-drivers/#downloading-a-module) carefully. You're not downloading the module, but the HTML page that links to it.

robertklep, 2025-07-24:
It's again showing `usbfs` now 🤔
```
|__3-1 10c4:ea60:0100 00 2.00 12MBit/s 100mA 1IF (Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 d88825e08fc2ef1184a8c6138148b910)
3-1:1.0 (IF) ff:00:00 2EPs () usbfs
```

Perhaps that's normal, I don't know.

You don't have any `/dev/ttyUSB*` or `/dev/ttyACM*` devices now, right?

rcork, 2025-07-24:
> I got SynoCli Kernel Tools and SynoKernel USB Serial drivers installed. You mean there no need to install dsm7-usb-serial-drivers manually anymore? Btw, what dsm is your ds1821+ nas running on?

I'm running DSM 7.2.2-72806 Update 3 on my 1821+. After i added the SynoCommunity to Package Center and then installed the SynoKernel USB Serial drivers package in Package Center and plugged in the usb device i needed serial drivers for (a FlipperZero that i can control via serial commands) the /dev/ttyACM0 directory was created and allowed me to control it. Interestingly, when i run lsmod, i dont see any serial driver loaded. But when i run lsusb -cui, i do see that cdc_acm is loaded

```
  |__1-2         0483:5740:0100 ef  2.00   12MBit/s 100mA 2IFs (Flipper Devices Inc. Flipper Ashaers flip_Ashaers)
  1-2:1.0         (IF) 02:02:00 1EP  () cdc_acm tty/ttyACM0 
  1-2:1.1         (IF) 0a:00:00 2EPs () cdc_acm 
```

robertklep, 2025-07-24:
@rcork I wonder if possibly the `cdc_acm` driver is compiled into the kernel instead of being provided as a module.

nvungoc, 2025-07-24:
> I'm running DSM 7.2.2-72806 Update 3 on my 1821+. After i added the SynoCommunity to Package Center and then installed the SynoKernel USB Serial drivers package in Package Center and plugged in the [...]

Interesting! cdc-acm is still missing in my system, but whatever, the main thing is the driver is working now as expected.

robertklep, 2025-07-24:
@nvungoc cool, thanks for letting us know. I'll close this issue for now as it appears that we have a solution.

robertklep, 2025-07-25:
@nutttz "Invalid module format" most likely means you didn't download the module files correctly, as explained [here](https://github.com/robertklep/dsm7-usb-serial-drivers/issues/126#issuecomment-3112294152).