robertklep/dsm7-usb-serial-drivers #127: DS1823xs+ - V1000 - PL2303GS -VID PID 067B 23A3

bibibi-Matrix, 2024-08-27:
Good evening. Please help, I can't connect the CH340 or PL2303GS UPS to my device - VID PID 067B 23A3. The device is DS1823xs+ - V1000 Synology DSM 7.2.1-69057 Update 5. Nothing helps and lsusb does not show pl2303. What can I saddle up? Thanks for your help

robertklep, 2024-08-28:
@bibibi-Matrix can you post the output of these two commands?

```
lsusb -cui
```

```
lsmod | grep serial
```

bibibi-Matrix, 2024-08-28:
> @bibibi-Matrix can you post the output of these two commands?
> 
> ```
> lsusb -cui
> ```
> 
> ```
> lsmod | grep serial
> ```

`lsusb -cui
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.2) hub
  |__3-2         f400:f400:1075 00  2.10  480MBit/s 200mA 1IF  (Jetflash Mass Storage Device 1674976972)
  3-2:1.0         (IF) 08:06:50 2EPs () usb-storage host10 (synoboot)
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.2) hub
  |__3-2         f400:f400:1075 00  2.10  480MBit/s 200mA 1IF  (Jetflash Mass Storage Device 1674976972)
  3-2:1.0         (IF) 08:06:50 2EPs () usb-storage host10 (synoboot)`

`lsmod | grep serial
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio`

Good morning. It feels like he doesn't see USB, but he turned on a USB flash drive in this port and it's working. PL2303 turned on the computer and it is also detected.

robertklep, 2024-08-28:
It should show up in the output of `lsusb`, even if there's no driver for it, so it seems to be a hardware issue (perhaps with the cable, sometimes it helps using a USB extension cable in between the NAS and your device).

bibibi-Matrix, 2024-08-30:
root@NAS-Matrix:/dev# lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:12.2) hub
  |__1-5         0bda:5411:0117 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-5.1     0bda:5411:0117 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
      |__1-5.1.1 067b:23a3:0605 00  2.00   12MBit/s 100mA 1IF  (Prolific Technology Inc.  USB-Serial Controller  CSAPb11BS14)
    1-5.1.1:1.0   (IF) ff:00:00 3EPs ()
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.2) hub
  |__3-2         f400:f400:0100 00  2.00  480MBit/s 100mA 1IF  (General  UDisk            Љ)
  3-2:1.0         (IF) 08:06:50 2EPs () usb-storage host10 (synoboot)

usbserial              22278  4 ch341,pl2303

bibibi-Matrix, 2024-08-30:
[87598.812628] pl2303 1-5.1.1:1.0: pl2303 converter detected
[87598.818223] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8484]: -32
[87598.826360] pl2303 1-5.1.1:1.0: pl2303_vendor_write - failed to write [0404]: -32
[87598.834214] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8484]: -32
[87598.842412] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8383]: -32
[87598.850214] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8484]: -32
[87598.858362] pl2303 1-5.1.1:1.0: pl2303_vendor_write - failed to write [0404]: -32
[87598.866278] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8484]: -32
[87598.874360] pl2303 1-5.1.1:1.0: pl2303_vendor_read - failed to read [8383]: -32
[87598.882370] pl2303 1-5.1.1:1.0: pl2303_vendor_write - failed to write [0000]: -32
[87598.890201] pl2303 1-5.1.1:1.0: pl2303_vendor_write - failed to write [0001]: -32
[87598.898271] pl2303 1-5.1.1:1.0: pl2303_vendor_write - failed to write [0002]: -32

In hex, I changed the VID and PID and, as you can see, it causes an error, but /dev/ttyusb1 created

robertklep, 2024-08-31:
It looks like it uses the CH341 driver, not the PL2303 driver 😊
```
[88238.318581] ch341 1-5.1.1:1.0: ch341-uart converter detected
[88238.326625] usb 1-5.1.1: ch341-uart converter now attached to ttyUSB1
```

I don't know how to configure Nut either, that's a bit beyond the scope of this repository.

bibibi-Matrix, 2024-08-31:
Because the UPS has ch340 and it started working, but pl2303 apparently needs to be rebuilt by adding vid and pid, but I do not know how to rebuild the module, in theory it is necessary to source the kernel and configs.

robertklep, 2024-08-31:
If the UPS has a CH340, why are you trying to get the PL2303 to work? Or do you need it for another device?

bibibi-Matrix, 2024-08-31:
And here everything is simple, the ch340 did not work for me right away and was not determined. I bought a pl2303 and it is also not determined, I have xpenology. And then I found out yesterday that the webcam is also not detected, but the flash drives work. Well, I plugged in the extension cord and nothing changed, and then I plugged in the USB hub and, lo and behold, everything was determined.

robertklep, 2024-08-31:
Good! I'll close this issue then 😊

closed 2024-08-31 by robertklep