robertklep/dsm7-usb-serial-drivers #128: Unknown Symbol in Module rndis_host.ko

lgriffin, 2024-08-29:
First off great work on getting these files in place, it is truly appreciated.

I am getting the following issue:
sudo insmod rndis_host.ko
insmod: ERROR: could not insert module rndis_host.ko: Unknown symbol in module

This is on the r1000/dsm-7.2 -- all other files installed without issue. Happy to try debug it on my end if you have any suggested steps?

robertklep, 2024-08-29:
Do you actually need this module?

lgriffin, 2024-08-29:
Right now I honestly don't know, I'm following a guide to get Zigbee up and running on my Synology and one of the docker containers keeps dying. I'm troubleshooting that at the moment but the only failed step was this particular module. Ironically my searching led me to #88 (which I didn't find earlier!) so there could be a genuine bug here for that module -- maybe not for my needs but for whomever that module was intended for

robertklep, 2024-08-30:
For Zigbee you won't need this driver 😊 It's usually `cdc-acm.ko` (which is shipped with DSM) or `ch341.ko`.

Do you know why the Docker container keeps dying? What kind of Zigbee dongle do you have?

lgriffin, 2024-09-05:
Sorry I haven't figured out the docker container yet, the dongle is the Sonoff Zigbee 3.0 USB Dongle Plus

robertklep, 2024-09-05:
That dongle won't require `rndis_host.ko`. I don't know specifically which one it _will_ require, but it should work just fine.