robertklep/dsm7-usb-serial-drivers #130: HP N54L baremetal Xpenology

gnat8, 2024-09-27:
Hi, 
I'm running Xpenology on HP N54L, platform broadwellnk/dsm-7.2
I'm trying to get UB ports working.
I followed the guide https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/ and then I follow up with manual install using the guide in this repository.
I manage to get working a memory USB stick (on three different usb ports out of 6 available) but the system does not recognise it everytime (using lsusb from terminal).
My real problem is to get working a serial USB adapter (succefully used on Raspberry and usually recognised as cp210x).
I plug it but it is not listed in lsusb.
FYI following your guide, after "/usr/local/etc/rc.d/usb-serial-drivers.sh start", I didn't get any reply from the terminal.
I'm newbie of linux console. I need some teach to do somekind of debug of what is happening.

Any chance to make it work?
Thank you!

robertklep, 2024-09-27:
A device not showing in the output of `lsusb` is a hardware problem. Try using a USB extension cable between your computer and the device.

gnat8, 2024-09-27:
Actually I have it on a 10 cm cable extension and I tried also with a passive USB hub. No difference.

robertklep, 2024-09-27:
Possibly something else then, but it has nothing to do with missing drivers I'm afraid.

gnat8, 2024-09-27:
[image]
Is it normal that only ch341 and pl2303 are loaded?

robertklep, 2024-09-28:
I would expect the cp210x driver to be loaded as well. You can manually load it to see why it won't load:
```
sudo insmod /lib/modules/cp210x.ko
```

However, the driver not being loaded has no bearing on the device not showing up in the output of `lsusb`, which is to show which USB devices are attached to your machine. A device needs to show up there before a driver be attached to it.

gnat8, 2024-09-28:
It replies "could not insert module, invalid module format".
In the meantime I'm trying different usb cable, powered hubs, bios option.
Many problems to be solved here..

robertklep, 2024-09-28:
Try downloading the module again, sounds like it didn't download correctly.

In any case, I can't fix hardware issues 🤷🏼‍♂️

gnat8, 2024-09-28:
[image]

Other than HW issue, I'm still not able to load the module.
I removed old cp210x.ko, and reloaded w/o positive result.
Everything done after 'sudo  su' command

robertklep, 2024-09-28:
You're not downloading from the correct URL, it should be this:
https://github.com/robertklep/dsm7-usb-serial-drivers/raw/refs/heads/main/modules/broadwellnk/dsm-7.2/cp210x.ko