robertklep/dsm7-usb-serial-drivers #131: Synology ds923+ and  Sonoff Zigbee 3.0 USB Dongle Plus

gawronix, 2024-12-11:
I did yors instruction without any problems to the point with command : lsusb -cui
Im not able to see the name of USB on ds923+ . Please help what you need more to help me resolve the issue ?

  |__1-3         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  1-3:1.0         (IF) ff:00:00 2EPs ()

when I'm trying to sudo chmod 666 /dev/ttyUSB0
chmod: cannot access '/dev/ttyUSB0': No such file or directory

The script is loading the correct RAW files upload and copied [cp210x.ko](https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/r1000/dsm-7.2/cp210x.ko)

WHAT IM MISSING ?!

gawronix, 2024-12-11:
:/$ /usr/syno/bin/lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:08:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650047DFB7238D86)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host7 (synoboot)
  |__1-3         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  1-3:1.0         (IF) ff:00:00 2EPs ()

robertklep, 2024-12-12:
What's the output of the command `lsmod | grep serial`?

gawronix, 2024-12-17:
[image]
sadly nothing :(

gawronix, 2024-12-17:
[image]

gawronix, 2024-12-17:
its just putty thing, im not including anything

gawronix, 2024-12-17:
[image]

robertklep, 2024-12-17:
> its just putty thing, im not including anything

Yes, you are. There's an extra `# ` before the command that needs to be removed:
[image]

The reason it was shown in the README is because by convention, it means that the command needs to be run as root. I've updated the README to remove any confusion.

gawronix, 2024-12-17:
Thanks. Now I need to get this work done in HA 👍

robertklep, 2024-12-18:
Good luck!

closed 2024-12-18 by robertklep