robertklep/dsm7-usb-serial-drivers #135: Can't find .ko files for DS923+ 7.2.1

avenirf, 2025-05-15:
**Request:** Prebuilt cp210x.ko and usbserial.ko  
**Platform:** r1000 (DS923+)  
**DSM version:** 7.2.1-72806  
**Kernel:** 4.4.302+  
Need drivers for Sonoff Zigbee 3.0 USB Dongle Plus-E (VID 10c4, PID ea60). Thanks!

robertklep, 2025-05-15:
R1000 drivers are [here](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/r1000/dsm-7.2).

`usbserial.ko` is not a driver I provide, it should come with DSM.

bfg100k, 2025-07-21:
> /usr/local/etc/rc.d/usb-serial-drivers.sh: line 7: `<!DOCTYPE html>'

This line clearly indicates that you did not manage to download the file from the repo properly (there are no HTML tags in the file usb-serial-drivers.sh). Likely the same issue for the ch341.ko file. Try using the "raw" links on github to download the files. Open the script file in a text editor to make sure your download was completed successfully. 

Lastly, you do not need the ch341.ko file for the Itead Sonoff Zigbee 3.0 USB Dongle Plus V2. The right serial driver for that device is cp210x.ko. You will still need generic ones though (i.e. usbserial.ko, ftdi_sio.ko, cdc-acm.ko)