robertklep/dsm7-usb-serial-drivers #137: Verzoek: usbserial.ko voor Avoton + DSM 7.1.1

robbertpaulussen, 2025-07-13:
Hoi Robert,

Ik heb een Synology DS1515+ (Avoton-platform) met DSM 7.1.1 en probeer via RS-232 mijn Victron Match UPS aan te sluiten met behulp van een USB-serieel converter.

Ik zie dat `ti_usb_3410_5052.ko` beschikbaar is voor deze combinatie, maar de benodigde `usbserial.ko` ontbreekt in de map voor DSM 7.1 / Avoton. Zou je deze module kunnen toevoegen?

Uitvoer van `uname -a`:

Linux MSL 3.10.108 #42962 SMP Mon Aug 19 15:12:09 CST 2024 x86_64 GNU/Linux synology_avoton_1515+

Alvast heel veel dank voor de moeite — en voor het beschikbaar stellen van je project. Het is enorm waardevol!

Groet,
Robbert

robertklep, 2025-07-14:
`usbserial.ko` zou standaard aanwezig moeten zijn (in `/lib/modules`) op DSM.

robbertpaulussen, 2025-07-14:
Dank voor je snelle antwoord, ik heb hierover heen gekeken, maar je hebt helemaal gelijk.

closed 2025-07-14 by robertklep